NVIDIA/yum-packaging-precompiled-kmod #12: Conflict with DKMS RPM

wpoely86, 2020-10-20:
The spec file (at least for RHEL 7) has:
```
Requires:               nvidia-driver-%{_named_version} = %{?epoch:%{epoch}:}%{kmod_driver_version}

Conflicts:      kmod-nvidia-latest-dkms
```

If I check for example `nvidia-driver-branch-450-450.80.02-1.el7.x86_64` (but it's not specific to this version):

```
rpm -qp nvidia-driver-branch-450-450.80.02-1.el7.x86_64.rpm --requires
...
kmod-nvidia-latest-dkms = 3:450.80.02
...
```

So the RPM on one hand has a conflict for `kmod-nvidia-latest-dkms` but it pulls in dependencies which requires `kmod-nvidia-latest-dkms`... How is this supposed to work?

wpoely86, 2020-10-20:
I've added a:

```
Provides:      kmod-nvidia-latest-dkms
```

and now it works. Is this the proper way to do it?

kmittman, 2020-10-20:
Hi @wpoely86, which "flavor" are you targeting? It looks like `branch-450` (precompiled), so I'm not sure why you would need to `Provides: kmod-nvidia-latest-dkms`.

The plugin, `yum-plugin-nvidia` (see https://github.com/NVIDIA/yum-packaging-nvidia-plugin or [RPM](https://developer.download.nvidia.com/compute/cuda/repos/rhel7/x86_64/yum-plugin-nvidia-0.5-1.el7.noarch.rpm) should filter the packages in the RPM repository based on your selection. Here is a slide showing the relationship for each flavor:
[image]

wpoely86, 2020-10-21:
OK, I missed the part about the yum plugin.

But I'm not a big fan of such an approach: using a plugin to alter the dependency tree. Why you use a 'provides' line like I suggested? That works too without needing a plugin?

kmittman, 2020-10-21:
What you mentioned in the first comment is the issue, `nvidia-driver-branch-450` having a `Requires: kmod-nvidia-latest-dkms = 3:450.80.02` is a bug. It should be `Requires: nvidia-kmod = 3:450.80.02`. For example, if you check the driver packages for [440.64.00](https://developer.download.nvidia.com/compute/cuda/repos/rhel7/x86_64/nvidia-driver-branch-440-440.64.00-1.el7.x86_64.rpm), it is as expected.

The regression was introduced when migrating to a new build system, which is setting `is_dkms` to true (passed as a parameter to the .spec file), for each of the 3 pipelines. Working on fixing that for future driver packages.

Regarding the rationale for the plugin, I've added some links to the bottom of the [README](https://github.com/NVIDIA/yum-packaging-precompiled-kmod#presentations) to help explain.

kmittman, 2020-10-22:
Okay I fixed the infrastructure issue, the packages generated now look like such

flavor: `branch-XXX`
**nvidia-driver-branch-450**-450.00.00-1.el7.x86_64.rpm
```shell
Requires: nvidia-kmod = 3:450.00.00
Provides: nvidia-kmod-common = 3:450.00.00
```
**kmod-nvidia-branch-450**-3.10.0-1160.r450.00.00.el7.x86_64.rpm
```shell
Provides: nvidia-kmod = 3:450.00.00
Provides: kmod-nvidia-branch-450 = 3:3.10.0-1160.r450.00.00.el7
Provides: kmod-nvidia-branch-450(x86-64) = 3:3.10.0-1160.r450.00.00.el7
```
---
flavor: `latest`
**nvidia-driver-latest**-450.00.00-1.el7.x86_64.rpm
```shell
Requires: nvidia-kmod = 3:450.00.00
Provides: nvidia-kmod-common = 3:450.00.00
```
**kmod-nvidia-latest**-3.10.0-1160.r450.00.00.el7.x86_64.rpm
```shell
Provides: nvidia-kmod = 3:450.00.00
Provides: kmod-nvidia-latest = 3:3.10.0-1160.r450.00.00.el7
Provides: kmod-nvidia-latest(x86-64) = 3:3.10.0-1160.r450.00.00.el7
```
---
flavor: `latest-dkms`
**nvidia-driver-latest-dkms**-450.00.00-1.el7.x86_64.rpm
```shell
Requires: kmod-nvidia-latest-dkms = 3:450.00.00
```
**kmod-nvidia-latest-dkms**-450.00.00-1.el7.x86_64.rpm
```shell
Provides: kmod-nvidia-latest-dkms = 3:450.00.00-1.el7
Provides: kmod-nvidia-latest-dkms(x86-64) = 3:450.00.00-1.el7
Provides: nvidia-kmod = 3:450.00.00
````

wpoely86, 2020-10-22:
In your kernel version, you're not using the full version like: `3.10.0-1127.19.1` but just `3.10.0-1127`. Is that deliberate?
The yum plugin depends on that?

kmittman, 2020-10-22:
Sorry if my example was misleading. You're right it does need to be the **full** driver string. Note that some RHEL kernels do have shorter strings:
```
3.10.0-1062.18.1.el7
3.10.0-1127.el7
3.10.0-1127.8.2.el7
3.10.0-1127.10.1.el7
3.10.0-1127.13.1.el7
3.10.0-1127.18.2.el7
3.10.0-1127.19.1.el7
3.10.0-1160.el7
3.10.0-1160.2.1.el7
3.10.0-1160.2.2.el7
```
Also note that passing the kernel string to the .spec file currently needs to be done in pieces, for example `3.10.0-1127.19.1.el7`
```
rpmbuild --define "%_topdir $(pwd)" --define "debug_package %{nil}" --define "kernel 3.10.0" --define "kernel_release 1127.19.1" --define "kernel_dist .el7" \
  --define "driver 450.80.02" --define "epoch 3" --define "driver_branch branch-450" -v -bb SPECS/yum-kmod-nvidia.spec
```
`kernel: 3.10.0` (the original upstream version)
`kernel_release: 1127.19.1` (indicates RHEL backports)
`kernel_dist: .el7` (distro extension; for RHEL8 kernels sometimes includes minor version like `.el8_2`)

kmittman, 2020-11-23:
This is fixed in `455.45.01` and future releases. Here is a table of RHEL7 driver packages indicating ones were built with the correct dependencies for precompiled.
| Branch  | Driver | Correct deps |
| ------------- | ------------- | ------------- |
| 418  | 418.87.00  | :heavy_check_mark:   |
| 418  | 418.87.01  | :heavy_check_mark:   |
| 418  | 418.116.00  | :heavy_check_mark:   |
| 418  | 418.126.02  | :heavy_check_mark:   |
| 418  | 418.152.00  | :x:    |
| 418  | 418.165.02  | :x:    |
| 440  | 440.33.01  | :heavy_check_mark:     |
| 440  | 440.64.00  | :heavy_check_mark:     |
| 440  | 440.95.01  | :x:    |
| 440  | 440.118.02  | :x:    |
| 450  | 450.36.06  | :x:    |
| 450  | 450.51.05  | :x:    |
| 450  | 450.51.06  | :x:    |
| 450  | 450.80.02  | :x:    |
| 455  | 455.23.05  | :x:    |
| 455  | 455.32.00  | :x:    |
| 455  | 455.45.01  | :heavy_check_mark:     |

closed 2020-11-23 by kmittman