NVIDIA/yum-packaging-precompiled-kmod #20: Trying to upgrade from 418 tries to pull in both 440 and 465 creating conflicts.

paulraines68, 2021-06-11:
On a machine that has not been upgraded in a long time I have installed from the yum repos:

```
[root@chewey ~]# rpm -qa | grep nvidia
nvidia-driver-devel-418.67-4.el7.x86_64
nvidia-driver-cuda-libs-418.67-4.el7.x86_64
nvidia-xconfig-418.67-1.el7.x86_64
nvidia-driver-NVML-418.67-4.el7.x86_64
nvidia-libXNVCtrl-devel-418.67-1.el7.x86_64
nvidia-libXNVCtrl-418.67-1.el7.x86_64
nvidia-detect-460.73.01-1.el7.elrepo.x86_64
nvidia-driver-NvFBCOpenGL-418.67-4.el7.x86_64
nvidia-modprobe-418.67-1.el7.x86_64
nvidia-driver-libs-418.67-4.el7.x86_64
yum-plugin-nvidia-0.5-1.el7.noarch
nvidia-settings-418.67-1.el7.x86_64
dkms-nvidia-418.67-1.el7.x86_64
nvidia-persistenced-418.67-1.el7.x86_64
nvidia-driver-418.67-4.el7.x86_64
nvidia-driver-cuda-418.67-4.el7.x86_64
```

Repo is defined as:

```
[root@chewey ~]# cat /etc/yum.repos.d/cuda.repo
[cuda]
name=cuda
baseurl=http://developer.download.nvidia.com/compute/cuda/repos/rhel7/x86_64
enabled=1
gpgcheck=1
gpgkey=http://developer.download.nvidia.com/compute/cuda/repos/rhel7/x86_64/7fa2af80.pub
```
When I do a yum upgrade it fails with:

```
--> Processing Conflict: 3:nvidia-driver-branch-440-440.118.02-1.el7.x86_64 conflicts nvidia-driver > 440.118.02
--> Processing Conflict: 3:nvidia-driver-latest-dkms-465.19.01-1.el7.x86_64 conflicts nvidia-driver > 465.19.01
--> Finished Dependency Resolution
Error: nvidia-driver-branch-440 conflicts with 3:nvidia-driver-latest-dkms-465.19.01-1.el7.x86_64
Error: Package: 3:nvidia-driver-branch-440-440.118.02-1.el7.x86_64 (cuda)
           Requires: kmod-nvidia-latest-dkms = 3:440.118.02
           Available: 3:kmod-nvidia-latest-dkms-418.87.00-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:418.87.00-1.el7
           Available: 3:kmod-nvidia-latest-dkms-418.87.00-2.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:418.87.00-2.el7
           Available: 3:kmod-nvidia-latest-dkms-418.87.01-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:418.87.01-1.el7
           Available: 3:kmod-nvidia-latest-dkms-418.116.00-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:418.116.00-1.el7
           Available: 3:kmod-nvidia-latest-dkms-418.126.02-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:418.126.02-1.el7
           Available: 3:kmod-nvidia-latest-dkms-418.152.00-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:418.152.00-1.el7
           Available: 3:kmod-nvidia-latest-dkms-418.165.02-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:418.165.02-1.el7
           Available: 3:kmod-nvidia-latest-dkms-418.181.07-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:418.181.07-1.el7
           Available: 3:kmod-nvidia-latest-dkms-418.197.02-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:418.197.02-1.el7
           Available: 3:kmod-nvidia-latest-dkms-440.33.01-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:440.33.01-1.el7
           Available: 3:kmod-nvidia-latest-dkms-440.64.00-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:440.64.00-1.el7
           Available: 3:kmod-nvidia-latest-dkms-440.95.01-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:440.95.01-1.el7
           Available: 3:kmod-nvidia-latest-dkms-440.118.02-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:440.118.02-1.el7
           Available: 3:kmod-nvidia-latest-dkms-450.36.06-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:450.36.06-1.el7
           Available: 3:kmod-nvidia-latest-dkms-450.51.05-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:450.51.05-1.el7
           Available: 3:kmod-nvidia-latest-dkms-450.51.06-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:450.51.06-1.el7
           Available: 3:kmod-nvidia-latest-dkms-450.80.02-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:450.80.02-1.el7
           Available: 3:kmod-nvidia-latest-dkms-450.102.04-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:450.102.04-1.el7
           Available: 3:kmod-nvidia-latest-dkms-450.119.03-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:450.119.03-1.el7
           Available: 3:kmod-nvidia-latest-dkms-450.119.04-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:450.119.04-1.el7
           Available: 3:kmod-nvidia-latest-dkms-455.23.05-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:455.23.05-1.el7
           Available: 3:kmod-nvidia-latest-dkms-455.32.00-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:455.32.00-1.el7
           Available: 3:kmod-nvidia-latest-dkms-455.45.01-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:455.45.01-1.el7
           Available: 3:kmod-nvidia-latest-dkms-460.27.04-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:460.27.04-1.el7
           Available: 3:kmod-nvidia-latest-dkms-460.32.03-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:460.32.03-1.el7
           Available: 3:kmod-nvidia-latest-dkms-460.73.01-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:460.73.01-1.el7
           Installing: 3:kmod-nvidia-latest-dkms-465.19.01-1.el7.x86_64 (cuda)
               kmod-nvidia-latest-dkms = 3:465.19.01-1.el7
Error: nvidia-driver-latest-dkms conflicts with 3:nvidia-driver-branch-440-440.118.02-1.el7.x86_64
 You could try using --skip-broken to work around the problem
```

Why is it trying to install both 440 and 465 driver files?

kmittman, 2021-06-11:
If I recall correctly, `418.87.00` (August 2019) was the first to use the "flavor" name scheme: https://github.com/NVIDIA/yum-packaging-nvidia-driver#deliverables
As such the [installation instructions](https://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html#redhat-installation) changed to
```
sudo yum install nvidia-driver-latest-dkms
```
This is needed to explicitly filter out the `latest` and `branch-XXX` packages (for precompiled).

As far as why `440` and `465`, I think that's just a result of the resolver trying to find an upgrade scenario that works. Might need to uninstall, as I don't recall if this upgrade is expected to install cleanly or not.

kmittman, 2021-06-14:
@paulraines68 please take a look and verify that these instructions solve the upgrade issue.

First I installed the old driver version using the local installer repo method
```shell
# Enable EPEL 7 repo
sudo yum install -y https://dl.fedoraproject.org/pub/epel/epel-release-latest-7.noarch.rpm
# Download NVIDIA driver local repo
wget http://us.download.nvidia.com/tesla/418.67/nvidia-diag-driver-local-repo-rhel7-418.67-1.0-1.x86_64.rpm
# Install/enable local repo
sudo rpm -i nvidia-diag-driver-local-repo-rhel7-418.67-1.0-1.x86_64.rpm
# Install NVIDIA driver packages (including nvidia-settings)
sudo yum install cuda-drivers
```

Very important after using a local repo
```shell
# Remove the local repo afterwards
sudo yum remove "*nvidia*driver-local-repo*"
```

Then as per the [CUDA Download Page](https://developer.nvidia.com/cuda-downloads?target_os=Linux&target_arch=x86_64&Distribution=RHEL&target_version=7&target_type=rpm_network),  enable the CUDA network repository
```shell
# Enable CUDA repo for RHEL7 (x86_64)
sudo yum-config-manager --add-repo https://developer.download.nvidia.com/compute/cuda/repos/rhel7/x86_64/cuda-rhel7.repo
# Force metadata to be refreshed
sudo yum clean all
```
If you try this (without nvidia-settings) it will fail to upgrade due to dependency conflicts
```shell
# Install the latest-dkms "flavor"
sudo yum install nvidia-driver-latest-dkms
```
```
Error: nvidia-driver-branch-440 conflicts with 3:nvidia-driver-latest-dkms-465.19.01-1.el7.x86_64
Error: Package: 3:nvidia-driver-branch-440-440.118.02-1.el7.x86_64 (cuda-rhel7-x86_64)
           Requires: kmod-nvidia-latest-dkms = 3:440.118.02
Error: nvidia-driver-latest-dkms conflicts with 3:nvidia-driver-branch-440-440.118.02-1.el7.x86_64
```

However, if you install the NVIDIA driver (`latest-dkms` flavor) **and** the cuda-drivers meta-package, then it will upgrade just fine to the latest version (`465.19`)
```shell
# Install the latest-dkms "flavor", plus nvidia-settings, nvidia-libXNVCtrl, nvidia-libXNVCtrl-devel
sudo yum install nvidia-driver-latest-dkms cuda-drivers
```
```
Installed:
  kmod-nvidia-latest-dkms.x86_64 3:465.19.01-1.el7              nvidia-driver-latest-dkms.x86_64 3:465.19.01-1.el7                 
  nvidia-driver-latest-dkms-NVML.x86_64 3:465.19.01-1.el7       nvidia-driver-latest-dkms-NvFBCOpenGL.x86_64 3:465.19.01-1.el7     
  nvidia-driver-latest-dkms-cuda.x86_64 3:465.19.01-1.el7       nvidia-driver-latest-dkms-cuda-libs.x86_64 3:465.19.01-1.el7       
  nvidia-driver-latest-dkms-devel.x86_64 3:465.19.01-1.el7      nvidia-driver-latest-dkms-libs.x86_64 3:465.19.01-1.el7            
  nvidia-libXNVCtrl.x86_64 3:465.19.01-1.el7                    nvidia-libXNVCtrl-devel.x86_64 3:465.19.01-1.el7                   
  nvidia-modprobe-latest-dkms.x86_64 3:465.19.01-1.el7          nvidia-persistenced-latest-dkms.x86_64 3:465.19.01-1.el7           
  nvidia-settings.x86_64 3:465.19.01-1.el7                      nvidia-xconfig-latest-dkms.x86_64 3:465.19.01-1.el7                

Dependency Installed:
  yum-plugin-nvidia.noarch 0:0.5-1.el7                                                                                              

Updated:
  cuda-drivers.x86_64 0:465.19.01-1                                                                                                 

Replaced:
  dkms-nvidia.x86_64 3:418.67-1.el7                             nvidia-driver.x86_64 3:418.67-4.el7                                
  nvidia-driver-NVML.x86_64 3:418.67-4.el7                      nvidia-driver-NvFBCOpenGL.x86_64 3:418.67-4.el7                    
  nvidia-driver-cuda.x86_64 3:418.67-4.el7                      nvidia-driver-cuda-libs.x86_64 3:418.67-4.el7                      
  nvidia-driver-devel.x86_64 3:418.67-4.el7                     nvidia-driver-libs.x86_64 3:418.67-4.el7                           
  nvidia-libXNVCtrl.x86_64 3:418.67-1.el7                       nvidia-libXNVCtrl-devel.x86_64 3:418.67-1.el7                      
  nvidia-modprobe.x86_64 3:418.67-1.el7                         nvidia-persistenced.x86_64 3:418.67-1.el7                          
  nvidia-settings.x86_64 3:418.67-1.el7                         nvidia-xconfig.x86_64 3:418.67-1.el7                               

Complete!
```

kmittman, 2021-06-28:
Closing, no response received after 2 weeks.

closed 2021-06-28 by kmittman