NVIDIA/yum-packaging-precompiled-kmod #24: Package kmod-nvidia-460.91.03-4.18.0-305.7.1-460.91.03-3.el8_4.x86_64.rpm is not signed

Dalesjo, 2021-07-22:
Hello

It is not possible to install all the branches with dnf because the signature is missing for some of the rpms. Trying to update driver with the following commands

```
dnf remove nvidia-driver
dnf module reset nvidia-driver
dnf module list nvidia-driver
dnf module install nvidia-driver:470
```

ends with error message

```
...
[SKIPPED] nvidia-driver-cuda-libs-470.57.02-1.el8.x86_64.rpm: Already downloaded
[SKIPPED] nvidia-driver-libs-470.57.02-1.el8.x86_64.rpm: Already downloaded
[SKIPPED] nvidia-libXNVCtrl-470.57.02-1.el8.x86_64.rpm: Already downloaded
[SKIPPED] nvidia-libXNVCtrl-devel-470.57.02-1.el8.x86_64.rpm: Already downloaded
Package kmod-nvidia-470.57.02-4.18.0-305.7.1-470.57.02-3.el8_4.x86_64.rpm is not signed
The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: GPG check FAILED
```

Running rpm -K $file on the latest kmod-nvidia for each branch shows this

```
kmod-nvidia-418.211.00-4.18.0-305.7.1-418.211.00-3.el8_4.x86_64.rpm: digests OK
kmod-nvidia-440.118.02-4.18.0-240.8.1-440.118.02-3.el8_3.x86_64.rpm: digests signatures OK
kmod-nvidia-450.142.00-4.18.0-305.7.1-450.142.00-3.el8_4.x86_64.rpm: digests OK
kmod-nvidia-455.45.01-4.18.0-305-455.45.01-3.el8.x86_64.rpm: digests signatures OK
kmod-nvidia-460.91.03-4.18.0-305.7.1-460.91.03-3.el8_4.x86_64.rpm: digests OK
kmod-nvidia-465.19.01-4.18.0-305.7.1-465.19.01-3.el8_4.x86_64.rpm: digests signatures OK
kmod-nvidia-470.57.02-4.18.0-305.7.1-470.57.02-3.el8_4.x86_64.rpm: digests OK
```

kmittman, 2021-07-22:
Thank you for reporting this @Dalesjo 
Yes, it's only the 4 precompiled kmod packages we shipped this afternoon.
This is now fixed (RPMs signed) and new kmod packages for the `4.18.0-305.10.2` kernel are [available](https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/precompiled/).
```shell
$ sudo dnf clean all                       
$ sudo dnf module install nvidia-driver:470
  kmod-nvidia-470.57.02-4.18.0-305.10.2       x86_64       3:470.57.02-3.el8_4     cuda-rhel8-x86_64        25 M
```

Dalesjo, 2021-07-22:
Can confirm it works now. Thank you for the quick fix.

closed 2021-07-22 by Dalesjo