NVIDIA/yum-packaging-precompiled-kmod #27: kmod-nvidia-470.57.02-4.18.0-305.17.1 for kernel version 4.18.0-305.17.1.el8_4 and NVIDIA driver 470.57.02 could be found

a4z, 2021-09-08:
following the recommendation in this discussion, 
https://access.redhat.com/solutions/4134401#comment-2124501
I am reporting this issue here since I just got

```
NOTE: Skipping kernel installation since no kernel module package kmod-nvidia-470.57.02-4.18.0-305.17.1 for kernel version 4.18.0-305.17.1.el8_4 and NVIDIA driver 470.57.02 could be found
Error: 
 Problem 1: package kernel-modules-4.18.0-305.17.1.el8_4.x86_64 requires kernel-uname-r = 4.18.0-305.17.1.el8_4.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-305.3.1.el8_4.x86_64
  - package kernel-core-4.18.0-305.17.1.el8_4.x86_64 is filtered out by exclude filtering
 Problem 2: package kernel-modules-4.18.0-305.17.1.el8_4.x86_64 requires kernel-uname-r = 4.18.0-305.17.1.el8_4.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-305.7.1.el8_4.x86_64
  - package kernel-core-4.18.0-305.17.1.el8_4.x86_64 is filtered out by exclude filtering
 Problem 3: package kernel-modules-4.18.0-305.17.1.el8_4.x86_64 requires kernel-uname-r = 4.18.0-305.17.1.el8_4.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-305.12.1.el8_4.x86_64
  - package kernel-core-4.18.0-305.17.1.el8_4.x86_64 is filtered out by exclude filtering
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)

``` 

which makes me sad :-(

kmittman, 2021-09-08:
Hi @a4z thank you for reporting this issue. It looks like RHEL8 kernel `4.18.0-305.17.1` was released yesterday afternoon but one of our CI/CD triggers failed. We are working to get this resolved by EOD, with compatible kmod packages for drivers  `470.57.02`, `460.91.03`, `450.142.00`, `418.211.00`.

kmittman, 2021-09-09:
Hey there, sorry for the delay, the packages are live now: https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/precompiled/
Please let me know if you have any issues.

a4z, 2021-09-09:
Did a `sudo dnf upgrade --refresh` and everything is fine now.

```
2021-09-09T18:29:08+0200 INFO Installed products updated.
2021-09-09T18:29:08+0200 DEBUG Upgraded: bpftool-4.18.0-305.17.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Upgraded: kernel-headers-4.18.0-305.17.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Upgraded: kernel-tools-4.18.0-305.17.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Upgraded: kernel-tools-libs-4.18.0-305.17.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Upgraded: python3-perf-4.18.0-305.17.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Installed: kernel-4.18.0-305.17.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Installed: kernel-core-4.18.0-305.17.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Installed: kernel-debug-core-4.18.0-305.17.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Installed: kernel-devel-4.18.0-305.17.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Installed: kernel-modules-4.18.0-305.17.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Installed: kmod-nvidia-470.57.02-4.18.0-305.17.1-3:470.57.02-3.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Removed: kernel-4.18.0-305.3.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Removed: kernel-core-4.18.0-305.3.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Removed: kernel-debug-core-4.18.0-240.22.1.el8_3.x86_64
2021-09-09T18:29:08+0200 DEBUG Removed: kernel-devel-4.18.0-305.3.1.el8_4.x86_64
2021-09-09T18:29:08+0200 DEBUG Removed: kernel-modules-4.18.0-305.3.1.el8_4.x86_64
2021-09-09T18:29:09+0200 INFO Complete!
2021-09-09T18:29:09+0200 DDEBUG Cleaning up.
```

Thanks a lot for the fast response and fix!

closed 2021-09-09 by a4z