NVIDIA/yum-packaging-precompiled-kmod #28: no kernel module package kmod-nvidia-470.57.02-4.18.0-305.19.1 for kernel version 4.18.0-305.19.1.el8_4 and NVIDIA driver 470.57.02 could be found

a4z, 2021-09-15:
It might be that I am a bit early, but maybe a CI machine has not act as it should, as in #27 so I post this already today

following the recommendation in this discussion,
https://access.redhat.com/solutions/4134401#comment-2124501
I am reporting this issue here since I just got

```
NOTE: Skipping kernel installation since no kernel module package kmod-nvidia-470.57.02-4.18.0-305.19.1 for kernel version 4.18.0-305.19.1.el8_4 and NVIDIA driver 470.57.02 could be found
Error: 
 Problem 1: package kernel-modules-4.18.0-305.19.1.el8_4.x86_64 requires kernel-uname-r = 4.18.0-305.19.1.el8_4.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-305.7.1.el8_4.x86_64
  - package kernel-core-4.18.0-305.19.1.el8_4.x86_64 is filtered out by exclude filtering
 Problem 2: package kernel-modules-4.18.0-305.19.1.el8_4.x86_64 requires kernel-uname-r = 4.18.0-305.19.1.el8_4.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-305.12.1.el8_4.x86_64
  - package kernel-core-4.18.0-305.19.1.el8_4.x86_64 is filtered out by exclude filtering
 Problem 3: package kernel-modules-4.18.0-305.19.1.el8_4.x86_64 requires kernel-uname-r = 4.18.0-305.19.1.el8_4.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-305.17.1.el8_4.x86_64
  - package kernel-core-4.18.0-305.19.1.el8_4.x86_64 is filtered out by exclude filtering
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
```

kmittman, 2021-09-16:
Hi Harald, the CI/CD pipelines are working through `4.18.0-305.19.1` but it's a multi-step process, with each validation step taking several hours to complete; for the four driver branches currently supported. This is why we give an estimate of about 24 hours turn-around time. I will update again, once it is posted (ETA ~ 06:00 UTC+2)

kmittman, 2021-09-16:
Okay kmod packages are live now: https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/precompiled/

Date | Component | Driver version | Kernel version | Kernel suffix
-- | -- | -- | -- | --
2021-09-15 | kmod-nvidia | 470.57.02 | 4.18.0-305.19.1 | el8_4.x86_64
2021-09-15 | kmod-nvidia | 460.91.03 | 4.18.0-305.19.1 | el8_4.x86_64
2021-09-15 | kmod-nvidia | 450.142.00 | 4.18.0-305.19.1 | el8_4.x86_64
2021-09-15 | kmod-nvidia | 418.211.00 | 4.18.0-305.19.1 | el8_4.x86_64

a4z, 2021-09-16:
Thanks you, I can confirm it works! 

Too sad that the experience RedHat + Nvidia is not as user friendly as it could be, like , say on Ubuntu. 
My T480 gets dated, and for the next Thinkpad I will either have to decide to skip the Nvidia card, or switch away from RH.
What is sad, since CentOS Stream looks actually appealing to me, but for Stream is no user friendly solution at all, or is it?

closed 2021-09-16 by a4z