NVIDIA/yum-packaging-precompiled-kmod #33: precompiled kmod packages unavailable for kernel 4.18.0-348.23.1

lamberss, 2022-05-06:
On my RHEL 8 machine when I try to update my machine, I get the following error:

```
Updating Subscription Management repositories.
Last metadata expiration check: 4:38:34 ago on Fri 06 May 2022 09:56:38 AM MDT.
NOTE: Skipping kernel installation since no kernel module package kmod-nvidia-510.47.03-4.18.0-348.23.1 for kernel version 4.18.0-348.23.1.el8_5 and NVIDIA driver 510.47.03 could be found
Error: 
 Problem: package kernel-modules-4.18.0-348.23.1.el8_5.x86_64 requires kernel-uname-r = 4.18.0-348.23.1.el8_5.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-348.20.1.el8_5.x86_64
  - package kernel-core-4.18.0-348.23.1.el8_5.x86_64 is filtered out by exclude filtering
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
```

I have been getting this message for over a week, which is longer than the ~24h that is advertised.  Is there an estimate for when the kmod will be available for RHEL 8 kernel 4.18.0-348.23.1

kmittman, 2022-05-06:
Hi @lamberss 
The [GPG key rotation](https://developer.nvidia.com/blog/updating-the-cuda-linux-gpg-repository-key/) that we did last week, has delayed new kmod RPM availability. My estimate would be [next week](https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/precompiled/).

kmittman, 2022-05-14:
Hi @lamberss 
Precompiled kmod packages are now available for the `4.18.0-372.9.1` kernel (RHEL 8.6 on May 10th)

Please note that to expedite this release, the previous (RHEL 8.5) kernel `4.18.0-348.23.1` was skipped. Let me know if you still need that version and we can go back to pick it up later.

michaelbarkdoll, 2022-05-16:
@kittman minor interest in 4.18.0-372.9.1, but it will resolve itself in a few days either way.

kmittman, 2022-05-31:
Hi @michaelbarkdoll 
Sorry I was on vacation, is there a combination still needed? The CI/CD appears to have posted a [few kmod packages](https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/precompiled/) in my absence.