NVIDIA/yum-packaging-precompiled-kmod #35: precompiled kmod packages unavailable for kernel 4.18.0-372.16.1

michaelbarkdoll, 2022-07-18:
Not sure why, but this one also appears to be delayed.  I had previous updated the GPG key:

# rpm -q gpg-pubkey --qf '%{NAME}-%{VERSION}-%{RELEASE}\t%{SUMMARY}\n'|grep nvidia
gpg-pubkey-d42d0685-62589a51	gpg(cudatools <cudatools@nvidia.com>)

# dnf update
Last metadata expiration check: 2:20:10 ago on Mon 18 Jul 2022 07:05:59 AM CDT.
NOTE: Skipping kernel installation since no kernel module package kmod-nvidia-515.48.07-4.18.0-372.16.1 for kernel version 4.18.0-372.16.1.el8_6 and NVIDIA driver 515.48.07 could be found
Error: 
 Problem 1: package kernel-modules-4.18.0-372.16.1.el8_6.x86_64 requires kernel-uname-r = 4.18.0-372.16.1.el8_6.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-348.20.1.el8_5.x86_64
  - package kernel-core-4.18.0-372.16.1.el8_6.x86_64 is filtered out by exclude filtering
 Problem 2: package kernel-modules-4.18.0-372.16.1.el8_6.x86_64 requires kernel-uname-r = 4.18.0-372.16.1.el8_6.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-372.9.1.el8.x86_64
  - package kernel-core-4.18.0-372.16.1.el8_6.x86_64 is filtered out by exclude filtering
 Problem 3: package kernel-modules-4.18.0-372.16.1.el8_6.x86_64 requires kernel-uname-r = 4.18.0-372.16.1.el8_6.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-372.13.1.el8_6.x86_64
  - package kernel-core-4.18.0-372.16.1.el8_6.x86_64 is filtered out by exclude filtering
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)

kmittman, 2022-07-18:
Hi @michaelbarkdoll 
I was OOTO much of last week, my colleague mentioned there was an error in our CI/CD automation (for building, testing, and releasing new kmod package combinations), looking into this now.

You can monitor the [status page](https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/precompiled/) for when we have kmod packages compiled for RHEL8 kernel `4.18.0-372.16.1`. I'll also update here when they are available for each supported driver branch.

turja16, 2022-07-23:
Any update on this issue?

kmittman, 2022-07-23:
Release testing completed successfully, it should be available tomorrow around 9am (UTC -0800).

kmittman, 2022-07-25:
I see the kmod package for `515.48.07` and kernel `4.18.0-372.16.1`
```
$ podman run -it rockylinux:8 /bin/bash -c "dnf install -y dnf-plugins-core sudo wget curl sudo; bash"
$ sudo dnf config-manager --add-repo https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/cuda-rhel8.repo
$ sudo dnf clean-all
$ sudo dnf module install nvidia-driver:515
```
```
 Installing dependencies:
  kmod-nvidia-515.48.07-4.18.0-372.16.1    x86_64    3:515.48.07-3.el8_6    cuda-rhel8-x86_64    30 M
```

```
$ dnf module info nvidia-driver:515
Name             : nvidia-driver
Stream           : 515 [e] [a]
Version          : 20220723080132
Context          : 0d422fd89a
Architecture     : x86_64
Profiles         : default [d] [i], fm, ks, src
Default profiles : default
Repo             : cuda-rhel8-x86_64
Summary          : Nvidia driver for 515 branch
Description      : This package provides the most recent NVIDIA display driver which allows for hardware accelerated rendering with recent NVIDIA chipsets.
                 : For the full product support list, please consult the release notes for driver version 515.48.07.
```

michaelbarkdoll, 2022-07-25:
Still trying to figure out exactly where it is messing up:
```
[root@localhost ~]# dnf clean all
170 files removed
[root@localhost ~]# dnf remove nvidia-driver
Dependencies resolved.
=====================================================================================================================================================================================================
 Package                                                         Architecture                     Version                                         Repository                                    Size
=====================================================================================================================================================================================================
Removing:
 kmod-nvidia-515.48.07-4.18.0-372.13.1                           x86_64                           3:515.48.07-3.el8_6                             @cuda-rhel8-x86_64                            52 M
 kmod-nvidia-515.48.07-4.18.0-372.9.1                            x86_64                           3:515.48.07-3.el8                               @cuda-rhel8-x86_64                            52 M
 nvidia-driver                                                   x86_64                           3:515.48.07-1.el8                               @cuda-rhel8-x86_64                            60 M
Removing dependent packages:
 cuda-drivers                                                    x86_64                           515.48.07-1                                     @cuda-rhel8-x86_64                             0  
 nvidia-kmod-common                                              noarch                           3:515.48.07-1.el8                               @cuda-rhel8-x86_64                           2.7 k
 nvidia-modprobe                                                 x86_64                           3:515.48.07-1.el8                               @cuda-rhel8-x86_64                            79 k
 nvidia-settings                                                 x86_64                           3:515.48.07-1.el8                               @cuda-rhel8-x86_64                           1.9 M
 nvidia-xconfig                                                  x86_64                           3:515.48.07-1.el8                               @cuda-rhel8-x86_64                           299 k
Removing unused dependencies:
 dnf-plugin-nvidia                                               noarch                           2.0-1.el8                                       @cuda-rhel8-x86_64                            13 k

Transaction Summary
=====================================================================================================================================================================================================
Remove  9 Packages

Freed space: 167 M
Is this ok [y/N]: y
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                                                                                             1/1 
  Running scriptlet: cuda-drivers-515.48.07-1.x86_64                                                                                                                                             1/1 
  Erasing          : cuda-drivers-515.48.07-1.x86_64                                                                                                                                             1/9 
  Erasing          : nvidia-modprobe-3:515.48.07-1.el8.x86_64                                                                                                                                    2/9 
  Erasing          : nvidia-settings-3:515.48.07-1.el8.x86_64                                                                                                                                    3/9 
  Erasing          : nvidia-xconfig-3:515.48.07-1.el8.x86_64                                                                                                                                     4/9 
  Running scriptlet: nvidia-driver-3:515.48.07-1.el8.x86_64                                                                                                                                      5/9 
  Erasing          : nvidia-driver-3:515.48.07-1.el8.x86_64                                                                                                                                      5/9 
  Running scriptlet: nvidia-driver-3:515.48.07-1.el8.x86_64                                                                                                                                      5/9 
  Running scriptlet: nvidia-kmod-common-3:515.48.07-1.el8.noarch                                                                                                                                 6/9 
  Erasing          : nvidia-kmod-common-3:515.48.07-1.el8.noarch                                                                                                                                 6/9 
  Running scriptlet: nvidia-kmod-common-3:515.48.07-1.el8.noarch                                                                                                                                 6/9 
[... cut ...]
                 : nvidia-persistenced-3:515.48.07-1.el8.x86_64
                 : nvidia-settings-3:515.43.04-1.el8.x86_64
                 : nvidia-settings-3:515.48.07-1.el8.x86_64
                 : nvidia-xconfig-3:515.43.04-1.el8.x86_64
                 : nvidia-xconfig-3:515.48.07-1.el8.x86_64

Hint: [d]efault, [e]nabled, [x]disabled, [i]nstalled, [a]ctive
[root@localhost ~]# 
[root@localhost ~]# uname -a
Linux localhost.fqdn.com 4.18.0-372.13.1.el8_6.x86_64 #1 SMP Wed Jun 29 17:21:09 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
[root@localhost ~]#
```

michaelbarkdoll, 2022-07-25:
Debating trying:

```
[root@localhost ~]# dnf info kmod-nvidia-515.48.07-4.18.0-372.16.1
Last metadata expiration check: 0:02:58 ago on Mon 25 Jul 2022 01:22:06 PM CDT.
NOTE: Skipping kernel installation since no kernel module package kmod-nvidia-515.48.07-4.18.0-372.16.1.el8_6.0 for kernel version 4.18.0-372.16.1.el8_6.0.1 and NVIDIA driver 515.48.07 could be found
Available Packages
Name         : kmod-nvidia-515.48.07-4.18.0-372.16.1
Epoch        : 3
Version      : 515.48.07
Release      : 3.el8_6
Architecture : x86_64
Size         : 30 M
Source       : kmod-nvidia-515.48.07-4.18.0-372.16.1-515.48.07-3.el8_6.src.rpm
Repository   : cuda-rhel8-x86_64
Summary      : NVIDIA graphics driver
URL          : http://www.nvidia.com/
License      : Nvidia
Description  : The NVidia 515.48.07 display driver kernel module for kernel 4.18.0-372.16.1.el8_6

[root@localhost ~]# dnf install kmod-nvidia-515.48.07-4.18.0-372.16.1
Last metadata expiration check: 0:10:26 ago on Mon 25 Jul 2022 01:22:06 PM CDT.
NOTE: Skipping kernel installation since no kernel module package kmod-nvidia-515.48.07-4.18.0-372.16.1.el8_6.0 for kernel version 4.18.0-372.16.1.el8_6.0.1 and NVIDIA driver 515.48.07 could be found
Dependencies resolved.
=====================================================================================================================================================================================================
 Package                                                         Architecture                     Version                                          Repository                                   Size
=====================================================================================================================================================================================================
Installing:
 kernel                                                          x86_64                           4.18.0-372.16.1.el8_6                            baseos                                      8.0 M
 kernel-core                                                     x86_64                           4.18.0-372.16.1.el8_6                            baseos                                       39 M
 kernel-modules                                                  x86_64                           4.18.0-372.16.1.el8_6                            baseos                                       32 M
 kmod-nvidia-515.48.07-4.18.0-372.16.1                           x86_64                           3:515.48.07-3.el8_6                              cuda-rhel8-x86_64                            30 M
Removing:
 kernel-core                                                     x86_64                           4.18.0-348.20.1.el8_5                            @baseos                                      68 M
Removing dependent packages:
 kernel                                                          x86_64                           4.18.0-348.20.1.el8_5                            @baseos                                       0  
 kernel-modules                                                  x86_64                           4.18.0-348.20.1.el8_5                            @baseos                                      22 M

Transaction Summary
=====================================================================================================================================================================================================
Install  4 Packages
Remove   3 Packages

Total download size: 109 M
Is this ok [y/N]: 
```

kmittman, 2022-07-25:
This is the latest kernel I see from `uname -r` is `4.18.0-372.16.1.el8_6.x86_64` but your output mentions: 
> for kernel version `4.18.0-372.16.1.el8_6.0.1` and NVIDIA driver `515.48.07` could be found.

Are you using the "vanilla" RHEL 8 kernel or on something like EUS ? 

Can you please share output from `uname -a`

```
$ uname -a
Linux dvssc-dci-vt011 4.18.0-372.16.1.el8_6.x86_64 #1 SMP Tue Jun 28 03:02:21 EDT 2022 x86_64 x86_64 x86_64 GNU/Linux
```
and RPM info from installed `kernel-core` package
```
$ rpm -qi $(rpm -qa | grep kernel-core | grep $(uname -r))
Name        : kernel-core
Version     : 4.18.0
Release     : 372.16.1.el8_6
Architecture: x86_64
Install Date: Wed 13 Jul 2022 06:00:43 AM EDT
Group       : System Environment/Kernel
Size        : 72036836
License     : GPLv2 and Redistributable, no modification permitted
Signature   : RSA/SHA256, Tue 28 Jun 2022 10:52:32 AM EDT, Key ID 199e2f91fd431d51
Source RPM  : kernel-4.18.0-372.16.1.el8_6.src.rpm
Build Date  : Tue 28 Jun 2022 03:36:45 AM EDT
Build Host  : x86-vm-08.build.eng.bos.redhat.com
Relocations : (not relocatable)
Packager    : Red Hat, Inc. <http://bugzilla.redhat.com/bugzilla>
Vendor      : Red Hat, Inc.
URL         : http://www.kernel.org/
Summary     : The Linux kernel
Description :
The kernel package contains the Linux kernel (vmlinuz), the core of any
Linux operating system.  The kernel handles the basic functions
of the operating system: memory allocation, process allocation, device
input and output, etc.
```

michaelbarkdoll, 2022-07-25:
I'm using Rocky Linux 8 which this has worked on in the past.
```
[root@localhost ~]# uname -a
Linux localhost.fqdn.edu 4.18.0-372.13.1.el8_6.x86_64 #1 SMP Wed Jun 29 17:21:09 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
[root@localhost ~]# rpm -qi $(rpm -qa | grep kernel-core | grep $(uname -r))
Name        : kernel-core
Version     : 4.18.0
Release     : 372.13.1.el8_6
Architecture: x86_64
Install Date: Sat 02 Jul 2022 06:49:50 AM CDT
Group       : System Environment/Kernel
Size        : 71989829
License     : GPLv2 and Redistributable, no modification permitted
Signature   : RSA/SHA256, Wed 29 Jun 2022 06:34:07 PM CDT, Key ID 15af5dac6d745a60
Source RPM  : kernel-4.18.0-372.13.1.el8_6.src.rpm
Build Date  : Wed 29 Jun 2022 12:46:16 PM CDT
Build Host  : dal1-prod-builder001.bld.equ.rockylinux.org
Relocations : (not relocatable)
Packager    : Release Engineering <releng@rockylinux.org>
Vendor      : Rocky
URL         : http://www.kernel.org/
Summary     : The Linux kernel
Description :
The kernel package contains the Linux kernel (vmlinuz), the core of any
Linux operating system.  The kernel handles the basic functions
of the operating system: memory allocation, process allocation, device
input and output, etc.
[root@localhost ~]# 
```

kmittman, 2022-07-25:
Ummm okay that didn't narrow it down, you are still on `4.18.0-372.13.1.el8_6`, I'd like to know what it is trying to upgrade to

Yeah there is a difference between available kernels for `rocky8` and `rhel8`: `4.18.0-372.16.1.el8_6.0.1`

```
[rocky8] $ dnf list kernel-core --showduplicates
Rocky Linux 8 - AppStream                       3.2 MB/s | 8.9 MB     00:02    
Rocky Linux 8 - BaseOS                                                                                              2.3 MB/s | 5.5 MB     00:02    
Rocky Linux 8 - Extras                                                                                               14 kB/s |  11 kB     00:00    
Available Packages
kernel-core.x86_64    4.18.0-372.9.1.el8            baseos
kernel-core.x86_64    4.18.0-372.13.1.el8_6         baseos
kernel-core.x86_64    4.18.0-372.16.1.el8_6         baseos
kernel-core.x86_64    4.18.0-372.16.1.el8_6.0.1     baseos
```
vs
```
[rhel8] $ dnf list kernel-core --showduplicates
Updating Subscription Management repositories.
Last metadata expiration check: 0:55:30 ago on Mon 25 Jul 2022 02:00:06 PM EDT.
Installed Packages
kernel-core.x86_64     4.18.0-372.9.1.el8         @rhel-8-for-x86_64-baseos-rpms
kernel-core.x86_64     4.18.0-372.13.1.el8_6      @rhel-8-for-x86_64-baseos-rpms
kernel-core.x86_64     4.18.0-372.16.1.el8_6      @rhel-8-for-x86_64-baseos-rpms
Available Packages
kernel-core.x86_64     4.18.0-80.el8              rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-80.1.2.el8_0        rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-80.4.2.el8_0        rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-80.7.1.el8_0        rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-80.7.2.el8_0        rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-80.11.1.el8_0       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-80.11.2.el8_0       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-147.el8             rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-147.0.2.el8_1       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-147.0.3.el8_1       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-147.3.1.el8_1       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-147.5.1.el8_1       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-147.8.1.el8_1       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-193.el8             rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-193.1.2.el8_2       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-193.6.3.el8_2       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-193.13.2.el8_2      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-193.14.3.el8_2      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-193.19.1.el8_2      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-193.28.1.el8_2      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-240.el8             rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-240.1.1.el8_3       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-240.8.1.el8_3       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-240.10.1.el8_3      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-240.15.1.el8_3      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-240.22.1.el8_3      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-305.el8             rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-305.3.1.el8_4       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-305.7.1.el8_4       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-305.10.2.el8_4      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-305.12.1.el8_4      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-305.17.1.el8_4      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-305.19.1.el8_4      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-305.25.1.el8_4      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-348.el8             rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-348.2.1.el8_5       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-348.7.1.el8_5       rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-348.12.2.el8_5      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-348.20.1.el8_5      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-348.23.1.el8_5      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-372.9.1.el8         rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-372.13.1.el8_6      rhel-8-for-x86_64-baseos-rpms 
kernel-core.x86_64     4.18.0-372.16.1.el8_6      rhel-8-for-x86_64-baseos-rpms 
```

michaelbarkdoll, 2022-07-25:
Odd, thanks for catching that.  Not sure why it doesn't show on my machine (the 4.18.0-372.16.1.el8_6.0.1).  Looks like this is a Rocky Linux issue in this case.

```
[root@localhost ~]# dnf list kernel-core --showduplicates
Last metadata expiration check: 0:21:49 ago on Mon 25 Jul 2022 01:40:34 PM CDT.
NOTE: Skipping kernel installation since no kernel module package kmod-nvidia-515.48.07-4.18.0-372.16.1.el8_6.0 for kernel version 4.18.0-372.16.1.el8_6.0.1 and NVIDIA driver 515.48.07 could be found
Installed Packages
kernel-core.x86_64                                                                           4.18.0-348.20.1.el8_5                                                                            @baseos
kernel-core.x86_64                                                                           4.18.0-372.9.1.el8                                                                               @baseos
kernel-core.x86_64                                                                           4.18.0-372.13.1.el8_6                                                                            @baseos
Available Packages
kernel-core.x86_64                                                                           4.18.0-372.9.1.el8                                                                               baseos 
kernel-core.x86_64                                                                           4.18.0-372.13.1.el8_6                                                                            baseos 
kernel-core.x86_64                                                                           4.18.0-372.16.1.el8_6                                                                            baseos 
[root@localhost ~]#
```

kmittman, 2022-07-25:
Ohhhh, okay now I understand the discrepancy. The `dnf-plugin-nvidia` Python plugin is removing that unsupported kernel from the `dnf` output. That is what is adding the warning notice at the top, "NOTE: Skipping kernel ...". I added detection to disable that skip if it is running in a container to make it easier to unit test but that made it less obvious.

And yes, it seems that Rocky Linux has a kernel version ahead of RHEL. Unfortunately, we only officially support RHEL kernels for this precompiled kmod project.

I think the workaround for now, would be to upgrade your kernel to `4.18.0-372.16.1.el8_6`
```
sudo dnf install kernel-core-4.18.0-372.16.1.el8_6
```
and then should be able to install the latest precompiled kmod
```
sudo dnf module install nvida-driver:latest
```

michaelbarkdoll, 2022-07-25:
Thanks, yes that works.

mstg, 2022-07-25:
Hi folks,

Just wanted to quickly jump in related to this issue. Current Rocky Linux kernel is not ahead of RHEL, but the `Release` tags can differ for a few reasons. Mostly, and in this case, it's because we've had to republish the same version because of a non-technical change. Because we never publish same NVR twice, we can add `.rocky` or `.0.1` in this case to the `Release` field.

`kernel-4.18.0-372.16.1.el8_6.0.1` and `kernel-4.18.0-372.16.1.el8_6` is in fact equal.

I'm not sure if there is a good way to make this work with kmods. Thankfully though we do keep all published artifacts so users can generally use the first released version using the `dnf install` method you provided above @kmittman.

Thanks,
Mustafa Gezen
Release Engineering lead @ Rocky Linux

kmittman, 2022-07-25:
Hi @mstg 
I'm going to close this issue but happy to continue discussing what you brought up in issue #36

closed 2022-07-25 by kmittman

michaelbarkdoll, 2022-07-26:
I realize the issue is much larger, so I don't want to distract too much from your efforts there so ignore this if needed.  However, I wanted to note the following command `sudo dnf install kernel-core-4.18.0-372.16.1.el8_6` succeeds to execute, but nvidia-smi is still failing to work after a reboot and dnf fails with:

```
4.18.0-372.13.1# dnf install kernel-core-4.18.0-372.16.1.el8_6
4.18.0-372.13.1# uname -r
4.18.0-372.13.1.el8_6.x86_64
4.18.0-372.13.1# dnf update
Last metadata expiration check: 0:00:57 ago on Tue 26 Jul 2022 12:07:53 PM CDT.
NOTE: Skipping kernel installation since no kernel module package kmod-nvidia-515.48.07-4.18.0-372.16.1.el8_6.0 for kernel version 4.18.0-372.16.1.el8_6.0.1 and NVIDIA driver 515.48.07 could be found
Error: 
 Problem 1: package kernel-modules-4.18.0-372.16.1.el8_6.0.1.x86_64 requires kernel-uname-r = 4.18.0-372.16.1.el8_6.0.1.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-372.9.1.el8.x86_64
  - package kernel-core-4.18.0-372.16.1.el8_6.0.1.x86_64 is filtered out by exclude filtering
 Problem 2: package kernel-modules-4.18.0-372.16.1.el8_6.0.1.x86_64 requires kernel-uname-r = 4.18.0-372.16.1.el8_6.0.1.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-372.13.1.el8_6.x86_64
  - package kernel-core-4.18.0-372.16.1.el8_6.0.1.x86_64 is filtered out by exclude filtering
 Problem 3: package kernel-modules-4.18.0-372.16.1.el8_6.0.1.x86_64 requires kernel-uname-r = 4.18.0-372.16.1.el8_6.0.1.x86_64, but none of the providers can be installed
  - cannot install the best update candidate for package kernel-modules-4.18.0-372.16.1.el8_6.x86_64
  - package kernel-core-4.18.0-372.16.1.el8_6.0.1.x86_64 is filtered out by exclude filtering
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
```
Looks like a dependency issue to me, which seems to be expected.

```
4.18.0-372.13.1# dnf nvidia-plugin
Last metadata expiration check: 0:01:37 ago on Tue 26 Jul 2022 12:07:53 PM CDT.

installed kernel: kernel-4.18.0-372.16.1.el8_6.x86_64

installed kmod(s): kmod-nvidia-515.48.07-4.18.0-372.13.1-3:515.48.07-3.el8_6.x86_64 kmod-nvidia-515.48.07-4.18.0-372.16.1-3:515.48.07-3.el8_6.x86_64 kmod-nvidia-515.48.07-4.18.0-372.9.1-3:515.48.07-3.el8.x86_64

available kernel(s): kernel-4.18.0-372.13.1.el8_6.x86_64 kernel-4.18.0-372.16.1.el8_6.0.1.x86_64 kernel-4.18.0-372.16.1.el8_6.x86_64 kernel-4.18.0-372.9.1.el8.x86_64

available kernel-core(s): kernel-core-4.18.0-372.13.1.el8_6.x86_64 kernel-core-4.18.0-372.16.1.el8_6.0.1.x86_64 kernel-core-4.18.0-372.16.1.el8_6.x86_64 kernel-core-4.18.0-372.9.1.el8.x86_64

available driver(s): nvidia-driver-3:515.43.04-1.el8.x86_64 nvidia-driver-3:515.48.07-1.el8.x86_64

available kmod(s): ...
```

On a separate machine, I'm trying to get nvidia-smi working with a prior kernel as well to avoid the issue.  It appears `dnf module install nvidia-driver:latest` also installs support for previously installed kernels.  However, when I boot to those older versions of the kernel the `nvidia-smi` command is failing there as well.  Would I need to do anything special to have an older kernel support with an installation of the latest nvidia-driver branch?

I tried to versionlock the old kernel prior to the install:
```
older_kernel# uname -r
4.18.0-372.9.1.el8.x86_64
```
```
older_kernel# dnf versionlock list
Last metadata expiration check: 1:33:29 ago on Tue 26 Jul 2022 10:34:08 AM CDT.
kernel-0:4.18.0-372.9.1.el8.*
kernel-headers-0:4.18.0-372.9.1.el8.*
kernel-tools-0:4.18.0-372.9.1.el8.*
kernel-tools-libs-0:4.18.0-372.9.1.el8.*
kernel-devel-0:4.18.0-372.9.1.el8.*
```
```
older_kernel# dnf info kernel-core
Last metadata expiration check: 1:42:04 ago on Tue 26 Jul 2022 10:34:08 AM CDT.
Installed Packages
Name         : kernel-core
Version      : 4.18.0
Release      : 372.16.1.el8_6
Architecture : x86_64
Size         : 69 M
Source       : kernel-4.18.0-372.16.1.el8_6.src.rpm
Repository   : @System
From repo    : baseos
Summary      : The Linux kernel
URL          : http://www.kernel.org/
[... 30 lines cut ...]
Summary      : The Linux kernel
URL          : http://www.kernel.org/
License      : GPLv2 and Redistributable, no modification permitted
Description  : The kernel package contains the Linux kernel (vmlinuz), the core of any
             : Linux operating system.  The kernel handles the basic functions
             : of the operating system: memory allocation, process allocation, device
             : input and output, etc.
```

```
older_kernel# dnf nvidia-plugin
Last metadata expiration check: 1:27:56 ago on Tue 26 Jul 2022 10:44:52 AM CDT.

installed kernel: kernel-4.18.0-372.16.1.el8_6.0.1.x86_64

installed kmod(s): kmod-nvidia-515.43.04-4.18.0-372.9.1-3:515.43.04-3.el8.x86_64 kmod-nvidia-515.48.07-4.18.0-372.13.1-3:515.48.07-3.el8_6.x86_64 kmod-nvidia-515.48.07-4.18.0-372.16.1-3:515.48.07-3.el8_6.x86_64 kmod-nvidia-515.48.07-4.18.0-372.9.1-3:515.48.07-3.el8.x86_64

available kernel(s): kernel-4.18.0-372.13.1.el8_6.x86_64 kernel-4.18.0-372.16.1.el8_6.0.1.x86_64 kernel-4.18.0-372.16.1.el8_6.x86_64 kernel-4.18.0-372.9.1.el8.x86_64

available kernel-core(s): kernel-core-4.18.0-372.13.1.el8_6.x86_64 kernel-core-4.18.0-372.16.1.el8_6.0.1.x86_64 kernel-core-4.18.0-372.16.1.el8_6.x86_64 kernel-core-4.18.0-372.9.1.el8.x86_64

available driver(s): nvidia-driver-3:515.43.04-1.el8.x86_64 nvidia-driver-3:515.48.07-1.el8.x86_64
```

Sorry, just have one particular end-user that had their machine off for the last few nvidia-driver releases and they don't have a kernel they can boot to do some machine learning programming.  Any tips are appreciated.

michaelbarkdoll, 2022-07-26:
```
older_kernel# rpm -qa |grep kernel
kernel-devel-4.18.0-372.16.1.el8_6.0.1.x86_64
kernel-core-4.18.0-372.9.1.el8.x86_64
kernel-tools-4.18.0-372.9.1.el8.x86_64
kernel-headers-4.18.0-372.9.1.el8.x86_64
kernel-modules-4.18.0-372.16.1.el8_6.x86_64
kernel-devel-4.18.0-372.16.1.el8_6.x86_64
kernel-4.18.0-372.16.1.el8_6.0.1.x86_64
kernel-devel-4.18.0-372.9.1.el8.x86_64
texlive-l3kernel-20180414-25.el8.noarch
kernel-core-4.18.0-372.16.1.el8_6.x86_64
kernel-core-4.18.0-372.16.1.el8_6.0.1.x86_64
kernel-debug-core-4.18.0-372.9.1.el8.x86_64
kernel-4.18.0-372.9.1.el8.x86_64
kernel-modules-4.18.0-372.9.1.el8.x86_64
kernel-4.18.0-372.16.1.el8_6.x86_64
kernel-modules-4.18.0-372.16.1.el8_6.0.1.x86_64
kernel-debug-core-4.18.0-372.16.1.el8_6.0.1.x86_64
kernel-debug-core-4.18.0-372.13.1.el8_6.x86_64
kernel-tools-libs-4.18.0-372.9.1.el8.x86_64
```

kmittman, 2022-09-20:
Okay please let me know if there are any remaining issues. Otherwise keeping this issue closed.