NVIDIA/yum-packaging-precompiled-kmod #37: RHEL7 inconsistency between the file mode of ld.gold.nvidia and the defattr

txangel, 2022-08-04:
In the RHEL7 branch the spec sets the mode of ld.gold.nvidia as 755
https://github.com/NVIDIA/yum-packaging-precompiled-kmod/blob/515-rhel7/kmod-nvidia.spec#L255

However the spec also sets defattr for files as 644
https://github.com/NVIDIA/yum-packaging-precompiled-kmod/blob/515-rhel7/kmod-nvidia.spec#L260

This causes `rpm -q --dump` report the mode as `0100644` while in reality the file is laid down when installed as 755, which in order causes integrity scanners warn about the inconsistency.

kmittman, 2022-08-04:
Okay, from http://ftp.rpm.org/max-rpm/s1-rpm-inside-files-list-directives.html#S3-RPM-INSIDE-FLIST-DEFATTR-DIRECTIVE

> `%defattr(<file mode>, <user>, <group>, <dir mode>)`
> `%attr(<mode>, <user>, <group>) file`

It seems the solution would be to add a `%attr` line immediately after `%defattr`
```
%files
%defattr(644,root,root,755)
%attr(755,-,-) %{_bindir}/ld.gold.nvidia.%{kmod_driver_version}/ld.gold
```

To match up with the system defaults, i.e.
```
$ ls -l /usr/bin/ld.gold 
-rwxr-xr-x. 1 root root /usr/bin/ld.gold
```

txangel, 2022-08-10:
thanks @kmittman, do you want us to file a PR with that proposal or will you instead own making the changes?

txangel, 2022-09-02:
@kmittman I went ahead and filed a PR with what is discussed here. 
Perhaps it needs to be a new branch 516? Depends on your release process.
I tested it locally an it seems to attempt adjusting the mode of the right file indeed.

kmittman, 2022-09-02:
Hi @txangel thank you for the pull request, can you please sign the commit(s), see: [CONTRIBUTING.md](https://github.com/NVIDIA/yum-packaging-precompiled-kmod/blob/main/CONTRIBUTING.md). Optionally, perhaps consider squashing the commits?

I can try to back-port the change but not sure what release(s) the fix would make it into, as there are cut-off dates for each branch.

txangel, 2022-09-06:
@kmittman I think the commits are squashed and the description is now signed according to the guideline (https://github.com/NVIDIA/yum-packaging-precompiled-kmod/pull/38).
Is that all good now?