NVIDIA/yum-packaging-precompiled-kmod #40: Help compiling latest nvidia package NVIDIA-Linux-x86_64-525.60.11.run

jnm27, 2022-12-14:
Hi,

Trying to run build.sh from main on NVIDIA-Linux-x86_64-525.60.11.run. It gets to the copy_rpms stage and fails.

==> copy_rpms(http://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64)
:: repodata/3690e13e1224a040cac6f8c9a3dbe96551659ec22cabee088ed0b7daa8b515cb-primary.xml.gz

ERROR: Unable to locate 525.60.11 driver packages in repository for rhel8/x86_64

Looks like only 525.60.13 is available in the cuda repository, but that version is not available as a run file.

kmittman, 2023-01-12:
Hi @jnm27 
Unfortunately not every released NVIDIA driver runfile is available in RPM format. 

Here are some paths to check for available runfile downloads:
https://github.com/NVIDIA/yum-packaging-precompiled-kmod#Download-a-NVIDIA-driver-runfile

For example:
* `525.60.13`: https://download.nvidia.com/XFree86/Linux-x86_64/525.60.13/
which is present in the CUDA repository and can be used to build precompiled kmod RPMs for custom kernels.

jnm27, 2023-01-13:
Hi.

Thanks. I have resorted to setting up our own pipeline for building from your git repos and run files from NVIDIA so that we have more flexibility.

closed 2023-01-13 by jnm27