NVIDIA/yum-packaging-precompiled-kmod #50: No Drivers for RHEL 8.10

Kydane, 2024-06-04:
I am currently on the latest precompiled kmod available for 8.9 that was released on 4/2

Kernel version: 4.18.0-513.24.1.el8_9.x86_64
Nvidia driver: 550.54.15

RHEL 8 has gone into maintenance mode with kernel version 4.18.0-553.el8_10 but the Nvidia drivers haven't quite caught up on the "latest" dnf module stream. 

`no kernel module package kmod-nvidia-555.42.02-4.18.0-553 for kernel version 4.18.0-553.el8_10 and NVIDIA driver 550.54.15 could be found`

Am I missing something?

Thanks!

michaelbarkdoll, 2024-06-06:
I'm also experiencing this issue.

hhue13, 2024-06-10:
Same here! The missing drivers are obviously blocking the upgrade from RHEL 8.9 to RHEL 8.10.

kmittman, 2024-06-11:
Resolved. There was a timing issue that impacted the precompiled kmod release pipelines
```
Date	        Component	Driver version	Kernel version	Kernel suffix
2024-06-06	kmod-nvidia	550.90.07	4.18.0-553.5.1	el8_10.x86_64
2024-06-05	kernel-core	-	        4.18.0-553.5.1	el8_10.x86_64
2024-06-05	dnf-plugin	2.1	        -	        el8.noarch
2024-06-04	kmod-nvidia	555.42.02       4.18.0-553.5.1	el8_10.x86_64
2024-06-04	kmod-nvidia	535.183.01      4.18.0-553.5.1	el8_10.x86_64
2024-06-04	kmod-nvidia	470.256.02      4.18.0-553.5.1	el8_10.x86_64
2024-05-22	kernel-core	-	        4.18.0-553	el8_10.x86_64
```

```
$ sudo dnf config-manager --add-repo https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/cuda-rhel8.repo
$ sudo dnf module list nvidia-driver

$ sudo dnf module install nvidia-driver:latest
Dependencies resolved.
====================================================================================================================================
 Package                                        Architecture     Version                          Repository                   Size
====================================================================================================================================
Installing group/module packages:
 cuda-drivers                                   x86_64           555.42.02-1                      cuda-rhel8-x86_64           8.1 k
 nvidia-driver                                  x86_64           3:555.42.02-1.el8                cuda-rhel8-x86_64           124 M
 nvidia-driver-NVML                             x86_64           3:555.42.02-1.el8                cuda-rhel8-x86_64           600 k
 nvidia-driver-NvFBCOpenGL                      x86_64           3:555.42.02-1.el8                cuda-rhel8-x86_64            77 k
 nvidia-driver-cuda                             x86_64           3:555.42.02-1.el8                cuda-rhel8-x86_64           530 k
 nvidia-driver-cuda-libs                        x86_64           3:555.42.02-1.el8                cuda-rhel8-x86_64            53 M
 nvidia-driver-devel                            x86_64           3:555.42.02-1.el8                cuda-rhel8-x86_64            13 k
 nvidia-driver-libs                             x86_64           3:555.42.02-1.el8                cuda-rhel8-x86_64           135 M
 nvidia-kmod-common                             noarch           3:555.42.02-1.el8                cuda-rhel8-x86_64            13 k
 nvidia-libXNVCtrl                              x86_64           3:555.42.02-1.el8                cuda-rhel8-x86_64            26 k
 nvidia-libXNVCtrl-devel                        x86_64           3:555.42.02-1.el8                cuda-rhel8-x86_64            56 k
 nvidia-modprobe                                x86_64           3:555.42.02-2.el8                cuda-rhel8-x86_64            37 k
 nvidia-persistenced                            x86_64           3:555.42.02-1.el8                cuda-rhel8-x86_64            45 k
 nvidia-settings                                x86_64           3:555.42.02-1.el8                cuda-rhel8-x86_64           841 k
 nvidia-xconfig                                 x86_64           3:555.42.02-2.el8                cuda-rhel8-x86_64           105 k
Installing dependencies:
 dnf-plugin-nvidia                              noarch           2.1-1.el8                        cuda-rhel8-x86_64            14 k
 kmod-nvidia-555.42.02-4.18.0-553.5.1           x86_64           3:555.42.02-3.el8_10             cuda-rhel8-x86_64            41 M
Installing module profiles:
 nvidia-driver/default                                                                                                             
Enabling module streams:
 nvidia-driver                                                   latest                                                            

Transaction Summary
====================================================================================================================================
Install  17 Packages

Total download size: 356 M
Installed size: 799 M
Is this ok [y/N]:  y
```

```
$ dnf nvidia-plugin

Installed kmod(s):
  kmod-nvidia-555.42.02-4.18.0-553.5.1-3:555.42.02-3.el8_10.x86_64

Available kernel(s):
  kernel-core-4.18.0-553.5.1.el8_10.x86_64
  kernel-core-4.18.0-553.el8_10.x86_64

Available driver(s):
  nvidia-driver-3:555.42.02-1.el8.x86_64
  nvidia-driver-cuda-3:555.42.02-1.el8.x86_64

Available kmod(s):
[...]
  kmod-nvidia-550.54.15-4.18.0-513.18.1-3:550.54.15-3.el8_9.x86_64
  kmod-nvidia-550.54.15-4.18.0-513.24.1-3:550.54.15-3.el8_9.x86_64
  kmod-nvidia-555.42.02-4.18.0-553.5.1-3:555.42.02-3.el8_10.x86_64
```

hhue13, 2024-06-13:
When I run my update RHEL8.10 with Kernel 4.18.0-553.5.1.el8_10.x86_64 I face the issue that the Laptop freezes if I don't have an external display attached.

hhue13, 2024-06-14:
Not sure if that's related but if fits the issue with the NVIDIA Driver on RHEL8 since RHEL 8.10 was released. Tried to run another `yum update` and here is what I get:

```
 sudo yum update
Updating Subscription Management repositories.
Last metadata expiration check: 11:09:29 ago on Thu 13 Jun 2024 08:17:15 PM CEST.
NVIDIA driver: filtering kernel 4.18.0-80.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.4.2.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.1.2.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.11.2.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.11.1.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.7.2.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.7.1.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.8.1.el8_1, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.5.1.el8_1, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.0.3.el8_1, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.3.1.el8_1, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.0.2.el8_1, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.1.2.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.19.1.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.6.3.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.13.2.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.14.3.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.28.1.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.1.1.el8_3, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.22.1.el8_3, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.8.1.el8_3, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.10.1.el8_3, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.15.1.el8_3, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.10.2.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.7.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.12.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.3.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.19.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.25.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.17.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.20.1.el8_5, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.7.1.el8_5, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.23.1.el8_5, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.2.1.el8_5, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.12.2.el8_5, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-372.9.1.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-425.3.1.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-372.16.1.el8_6, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-372.32.1.el8_6, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-372.13.1.el8_6, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-372.19.1.el8_6, no precompiled modules available for version 3:555.42.02
[... cut ...]
NVIDIA driver: filtering kernel 4.18.0-513.24.1.el8_9, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-553.el8_10, no precompiled modules available for version 3:555.42.02
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/dnf/plugin.py", line 107, in _caller
    getattr(plugin, method)()
  File "/usr/lib/python3.6/site-packages/dnf-plugins/nvidia.py", line 155, in resolved
    if pkg.name == DRIVER_PKG_NAME:
NameError: name 'DRIVER_PKG_NAME' is not defined
```

scaronni-nvidia, 2024-06-14:
Yes the python exception is fixed in 2.1-2.el8. The issues you are seeing is that the precompiled branch 555 is missing all precompiled kernel module for any version except for the combination **driver 555.42.02 + kernel 4.18.0-553.5.1**, which I see missing in your list, you stop at **553**.

We are looking into it.

scaronni, 2024-06-17:
A precompiled kernel module package has been posted for driver **555.42.02** and kernel **4.18.0-553.5.1**.

kmittman, 2024-06-17:
Thank you @scaronni, closing.

hhue13, 2024-06-18:
Thanks but I'm still getting the same issue:

```
[me@myhost ~]$ sudo yum update
Updating Subscription Management repositories.
Last metadata expiration check: 0:02:11 ago on Tue 18 Jun 2024 06:59:39 AM CEST.
NVIDIA driver: filtering kernel 4.18.0-80.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.4.2.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.1.2.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.11.2.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.11.1.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.7.2.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-80.7.1.el8_0, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.8.1.el8_1, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.5.1.el8_1, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.0.3.el8_1, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.3.1.el8_1, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-147.0.2.el8_1, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.1.2.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.19.1.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.6.3.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.13.2.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.14.3.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-193.28.1.el8_2, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.1.1.el8_3, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.22.1.el8_3, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.8.1.el8_3, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.10.1.el8_3, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-240.15.1.el8_3, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.10.2.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.7.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.12.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.3.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.19.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.25.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-305.17.1.el8_4, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.20.1.el8_5, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.7.1.el8_5, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.23.1.el8_5, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.2.1.el8_5, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-348.12.2.el8_5, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-372.9.1.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-425.3.1.el8, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-372.16.1.el8_6, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-372.32.1.el8_6, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-372.13.1.el8_6, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-372.19.1.el8_6, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-372.26.1.el8_6, no precompiled modules available for version 3:555.42.02
[... cut ...]
NVIDIA driver: filtering kernel 4.18.0-553.el8_10, no precompiled modules available for version 3:555.42.02
NVIDIA driver: filtering kernel 4.18.0-553.5.1.el8_10, no precompiled modules available for version 3:555.42.02
Dependencies resolved.
Nothing to do.
Complete!
[me@myhost ~]$ uname -a
Linux myhost.hhue.at 4.18.0-553.5.1.el8_10.x86_64 #1 SMP Tue May 21 03:13:04 EDT 2024 x86_64 x86_64 x86_64 GNU/Linux
```
when I try to run a yum update. Anything I'm missing to get the updated kernel module package?

chindokae, 2024-06-18:
Same here.  I got 47 new files but there were no kmods.  The new files were  libvinfer*, python-libvinfer*, libvonnxparsers*, tensorrt*, and a new dnf-plugin.

chindokae, 2024-06-18:
This is from on of my CUDA dev machines.  They are behind corporate firewalls so the repos are air-gapped, hence the non-standard names.  We are stuck on kernel 4.18.0-513.24.1 because the update to 4.18.0-553.5.1 remains blocked by the lack of precompiled modules.  We did get a new dnf-plugin on yesterday's reposync but we did not get anything that looked like kmods.

NVIDIA driver: filtering kernel 4.18.0-147.0.2.el8_1, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-147.0.3.el8_1, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-147.3.1.el8_1, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-147.5.1.el8_1, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-147.8.1.el8_1, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-147.el8, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-193.1.2.el8_2, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-193.13.2.el8_2, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-193.14.3.el8_2, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-193.19.1.el8_2, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-193.28.1.el8_2, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-193.6.3.el8_2, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-193.el8, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-240.1.1.el8_3, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-240.10.1.el8_3, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-240.15.1.el8_3, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-240.22.1.el8_3, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-240.8.1.el8_3, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-240.el8, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-305.10.2.el8_4, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-305.12.1.el8_4, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-305.17.1.el8_4, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-305.19.1.el8_4, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-305.25.1.el8_4, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-305.3.1.el8_4, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-305.7.1.el8_4, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-305.el8, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-348.12.2.el8_5, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-348.2.1.el8_5, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-348.20.1.el8_5, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-348.23.1.el8_5, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-348.7.1.el8_5, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-348.el8, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-372.13.1.el8_6, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-372.16.1.el8_6, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-372.19.1.el8_6, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-372.26.1.el8_6, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-372.32.1.el8_6, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-372.9.1.el8, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-425.10.1.el8_7, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-425.13.1.el8_7, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-425.19.2.el8_7, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-425.3.1.el8, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-477.10.1.el8_8, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-477.13.1.el8_8, no precompiled modules available for version 3:550.54.15
NVIDIA driver: filtering kernel 4.18.0-477.15.1.el8_8, no precompiled modules available for version 3:550.54.15
[... cut ...]
  - package kmod-nvidia-555.42.02-4.18.0-553.5.1-3:555.42.02-3.el8_10.x86_64 from LOCAL-SERVER-CUDA-REPO is filtered out by modular filtering
  - package kmod-nvidia-open-dkms-3:555.42.02-1.el8.x86_64 from LOCAL-SERVER-CUDA-REPO is filtered out by modular filtering
(try to add '--allowerasing' to command line to replace conflicting packages or '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
andidate packages)
# uname -a Linux cuda.local 4.18.0-513.24.1.el8_9.x86_64 #1 SMP Thu Mar 14 14:20:09 EDT 2024 x86_64 x86_64 x86_64 GNU/Linux

scaronni-nvidia, 2024-06-18:
Hi, yes, the combination **550.54.15** and kernel **4.18.0-553.5.1.el8_10** is missing, but is available for driver version **550.90.07**:

https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/kmod-nvidia-550.90.07-4.18.0-553.5.1-550.90.07-3.el8_10.x86_64.rpm

It should appear inside the appropriate stream `nvidia-driver:550`.

hhue13, 2024-06-18:
Sorry but I'm a bit lost now. How can I get the correct driver installed then? So far all I had to to was to run a `yum update` and things got installed on RHEL8. 

That's the repo configured on my RHEL 8_10 system:
```
[cuda-rhel8-x86_64]
name=cuda-rhel8-x86_64
baseurl=https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64
enabled=1
gpgcheck=1
gpgkey=https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/D42D0685.pub
```
Do I have to modify anything here?

hhue13, 2024-06-18:
> Hi, yes, the combination **550.54.15** and kernel **4.18.0-553.5.1.el8_10** is missing, but is available for driver version **550.90.07**: [...]

After installing the rpm and booting using kernel `Linux xxx.hhue.at 4.18.0-553.5.1.el8_10.x86_64 #1 SMP Tue May 21 03:13:04 EDT 2024 x86_64 x86_64 x86_64 GNU/Linux` GNOME did not start at all anymore. So not sure if the module works.

chindokae, 2024-06-18:
There is a conflict between Redhat's kmod-kvdo package and it's available kernel modules.  kmod-kvdo needs to be removed.   After that, yum update still requires --allow-erasing to proceed with patching.  Even after all that, the kernel stays at  4.18.0-513.24.1.  

Once the patching finished and I rebooted, I was able to reinstall kmod-kvdo and that installed kernel 4.18.0-553.5.1.

The kmod-kvdo conflict only exists on machines with existing installations of NVIDIA drivers from this repo.  All the other machines without NV drivers patched without error.

chindokae, 2024-06-18:
And that update failed to install with the following error:  FATAL : modpost : GPL-incompatible module nvidia.ko uses GPL-only symbol 'lock_is_held_type  in make.log.

scaronni-nvidia, 2024-06-18:
> After installing the rpm and booting using kernel `Linux xxx.hhue.at 4.18.0-553.5.1.el8_10.x86_64 #1 SMP Tue May 21 03:13:04 EDT 2024 x86_64 x86_64 x86_64 GNU/Linux` GNOME did not start at all anymore. So not sure if the module works.

Can you check if the module is loaded? From the screen, Ctrl+Alt+F4 (for example) or via ssh.

scaronni-nvidia, 2024-06-18:
> There is a conflict between Redhat's kmod-kvdo package and it's available kernel modules. kmod-kvdo needs to be removed. After that, yum update still requires --allow-erasing to proceed with patching. Even after all that, the kernel stays at 4.18.0-513.24.1.

Kernel version 4.18.0-513.24.1 has modules only for specific versions:

```
kmod-nvidia-525.147.05-4.18.0-513.24.1-525.147.05-3.el8_9.x86_64.rpm
kmod-nvidia-535.161.08-4.18.0-513.24.1-535.161.08-3.el8_9.x86_64.rpm
kmod-nvidia-545.23.08-4.18.0-513.24.1-545.23.08-3.el8_9.x86_64.rpm
kmod-nvidia-550.54.15-4.18.0-513.24.1-550.54.15-3.el8_9.x86_64.rpm

> Once the patching finished and I rebooted, I was able to reinstall kmod-kvdo and that installed kernel 4.18.0-553.5.1.
> The kmod-kvdo conflict only exists on machines with existing installations of NVIDIA drivers from this repo. All the other machines without NV drivers patched without error.

Yes, because unfortunately we can not sign and ship kernel modules at the same time that Red Hat ships a new kernel + kmod-vdo package with signed kernel modules in it.

scaronni-nvidia, 2024-06-18:
@chindokae what is your target setup? 550 + latest kernel? latest driver regardless of branch (555)?

hhue13, 2024-06-18:
> Can you check if the module is loaded? From the screen, Ctrl+Alt+F4 (for example) or via ssh.

```
cat lsmod.out | grep nvi
nvidia_drm             90112  0
nvidia_modeset       1294336  1 nvidia_drm
nvidia_uvm           4567040  0
nvidia              53403648  2 nvidia_uvm,nvidia_modeset
drm_kms_helper        184320  3 drm_display_helper,nvidia_drm,i915
drm                   602112  8 drm_kms_helper,drm_display_helper,nvidia,drm_buddy,nvidia_drm,i915,ttm
video                  57344  3 thinkpad_acpi,i915,nvidia_modeset
```

Furthermore I'm attachng the lsmod output as well. 

[lsmod.out.gz](https://github.com/user-attachments/files/15890072/lsmod.out.gz)

What are the steps required to reinstall kmod-kvdo? Is that really required?

scaronni-nvidia, 2024-06-18:
> What are the steps required to reinstall kmod-kvdo? Is that really required?

I guess that if you ask you don't need it? It's for deduplicating logical volumes. For RHEL 10 it will be part of the kernel: https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/9/html-single/deduplicating_and_compressing_logical_volumes_on_rhel/index

scaronni-nvidia, 2024-06-18:
@chindokae @hhue13 we found an issue with the repository metadata, the `nvidia-driver:latest` stream is missing the kernel modules that are in `nvidia-driver:555`. Let's check again after we regenerate the metadata.

scaronni-nvidia, 2024-06-18:
> Updating Subscription Management repositories.
> Last metadata expiration check: 11:09:29 ago on Thu 13 Jun 2024 08:17:15 PM CEST.
> NVIDIA driver: filtering kernel 4.18.0-80.el8, no precompiled modules [...]

I will change the output of the plugin as a separate update, all these lines if you use Satellite and have all the historical content is too much. I think a one line notice and a suggestion to run `dnf nvidia-plugin` should be enough.

kmittman, 2024-06-18:
I've rebuilt the RPM repo metadata for `rhel8/x86_64`, please check if this solves the issues mentioned.
For RHEL9, I've opened #51

chindokae, 2024-06-19:
I built a clean RHEL8.10 machine with a 1080TI in it.  I patched it once to kernel  4.18.0-553.5.1 and rebooted.  

Since it was a clean install there was no preexisting NV driver and this should be the least complex install possible.

This was built on the open Internet.  No repo mirroring.  No security policy. 

[root@r101 user]# uname -a
Linux r101.home.local 4.18.0-553.5.1.el8_10.x86_64 #1 SMP Tue May 21 03:13:04 EDT 2024 x86_64 x86_64 x86_64 GNU/Linux

[root@r101 scripts]# yum update
Updating Subscription Management repositories.
Last metadata expiration check: 0:18:24 ago on Tue 18 Jun 2024 09:22:48 PM EDT.
Dependencies resolved.
Nothing to do.
Complete!

I then ran:

sudo dnf config-manager --add-repo https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/cuda-rhel8.repo

Did a yum update, got the lists.  Then I tried running "yum install nvidia-driver":

[root@r101 scripts]# yum install nvidia-driver
Updating Subscription Management repositories.
Last metadata expiration check: 0:16:08 ago on Tue 18 Jun 2024 09:22:48 PM EDT.
Error:
 Problem: package nvidia-driver-3:555.42.02-1.el8.x86_64 from cuda-rhel8-x86_64 requires nvidia-kmod-common = 3:555.42.02, but none of the providers can be installed
  - package nvidia-kmod-common-3:555.42.02-1.el8.noarch from cuda-rhel8-x86_64 requires nvidia-kmod = 3:555.42.02, but none of the providers can be installed
  - conflicting requests
  - package kmod-nvidia-555.42.02-4.18.0-553.5.1-3:555.42.02-3.el8_10.x86_64 from cuda-rhel8-x86_64 is filtered out by modular filtering
  - nothing provides dkms needed by kmod-nvidia-latest-dkms-3:555.42.02-1.el8.x86_64 from cuda-rhel8-x86_64
  - package kmod-nvidia-open-dkms-3:555.42.02-1.el8.x86_64 from cuda-rhel8-x86_64 is filtered out by modular filtering
  - nothing provides dkms needed by kmod-nvidia-open-dkms-3:555.42.02-1.el8.x86_64 from cuda-rhel8-x86_64
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)

These are the nvidia-driver packages yum knows about:

[root@r101 scripts]# yum search nvidia-driver
Updating Subscription Management repositories.
Last metadata expiration check: 0:22:33 ago on Tue 18 Jun 2024 09:22:48 PM EDT.
================================================================ Name Exactly Matched: nvidia-driver =================================================================
nvidia-driver.x86_64 : NVIDIA's proprietary display driver for NVIDIA graphic cards
=============================================================== Name & Summary Matched: nvidia-driver ================================================================
nvidia-driver-cuda.x86_64 : CUDA integration for nvidia-driver
nvidia-driver-cuda-libs.i686 : Libraries for nvidia-driver-cuda
nvidia-driver-cuda-libs.x86_64 : Libraries for nvidia-driver-cuda
nvidia-driver-devel.i686 : Development files for nvidia-driver
nvidia-driver-devel.x86_64 : Development files for nvidia-driver
nvidia-driver-libs.i686 : Libraries for nvidia-driver
nvidia-driver-libs.x86_64 : Libraries for nvidia-driver
==================================================================== Name Matched: nvidia-driver =====================================================================
nvidia-driver-NVML.i686 : NVIDIA Management Library (NVML)
nvidia-driver-NVML.x86_64 : NVIDIA Management Library (NVML)
nvidia-driver-NvFBCOpenGL.i686 : NVIDIA OpenGL-based Framebuffer Capture libraries
nvidia-driver-NvFBCOpenGL.x86_64 : NVIDIA OpenGL-based Framebuffer Capture libraries

This is all I've ever needed to do to get a driver installed from this repo.

scaronni-nvidia, 2024-06-19:
> I ran dnf module install nvidia-driver:latest-dkms and it worked without errors.

Just did a test and I could install, upgrade, switch branch to `nvidia-driver:latest` and upgrade, etc on both el8 and el9 with the latest metadata, so the issue of the missing precompiled kernel module in the `latest` stream is fixed.

hhue13, 2024-06-19:
> I've rebuilt the RPM repo metadata for `rhel8/x86_64`, please check if this solves the issues mentioned. For RHEL9, I've opened #51

No change for me :(

scaronni, 2024-06-19:
@hhue13 This issue is for dependencies and not being able to resolve prcompiled modules via DNF on module streams. Fi your original comment at  https://github.com/NVIDIA/yum-packaging-precompiled-kmod/issues/50#issuecomment-2158205724 is fixed, please open a new issue if your problem pertains (like it seems) about a display not working, etc.? 

Also please post module version and driver version or the nvidia bug report. Based on comment https://github.com/NVIDIA/yum-packaging-precompiled-kmod/issues/50#issuecomment-2177708213 it seems you are running the drivers on an Optimus laptop, so the display should be driven by your Intel card.

hhue13, 2024-06-19:
> @hhue13 This issue is for dependencies and not being able to resolve prcompiled modules via DNF on module streams. Fi your original comment at [#50 (comment)](https://github.com/NVIDIA/yum-packaging-precompiled-kmod/issues/50#issuecomment-2158205724) is fixed, please open a new issue if your problem pertains (like it seems) about a display not working, etc.?
> 
> Also please post module version and driver version or the nvidia bug report. Based on comment [#50 (comment)](https://github.com/NVIDIA/yum-packaging-precompiled-kmod/issues/50#issuecomment-2177708213) it seems you are running the drivers on an Optimus laptop, so the display should be driven by your Intel card.

I'm running on a Lenovo P16 Gen1 Laptop and that used to work without any issues for over 2 years and broke only with the RHEL 8_10 Upgrade. Since then I can't get this working. 

Running `dnf module install nvidia-driver:latest-dkms` results in the following error:
```
Error: 
 Problem: package kmod-nvidia-555.42.02-4.18.0-553.5.1-3:555.42.02-3.el8_10.x86_64 from @System conflicts with kmod-nvidia-latest-dkms provided by kmod-nvidia-latest-dkms-3:555.42.02-1.el8.x86_64 from cuda-rhel8-x86_64
  - conflicting requests
  - problem with installed package kmod-nvidia-555.42.02-4.18.0-553.5.1-3:555.42.02-3.el8_10.x86_64
(try to add '--allowerasing' to command line to replace conflicting packages or '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
```
What changed with the RHEL8_10 upgrade so that this suddenly does not work anymore?

chindokae, 2024-06-20:
@hhue13   At this point I would try removing the NVIDIA packages, and if you have it, the kmod-kvdo packages as well.   On my test machine removing  the kmod-kvdo driver allowed normal patching to occur after it rebuilt the boot initrd ramdisk.   In my case at least, I think kmod-kvdo caused the entire mess.  The NV kmods evidently had issues as well but I never go to the point of doing an NV kmod install because the kmod-kvdo kernel mod issue was blocking any further installation of kernel modules.   I don't run Ceph or host VMs with virtual drives, so I never even needed the virtual drive optimizer modules to begin with. 

It sounds like your machine has gotten into a state that may not be recoverable, though.   If the de-installation routines get confused by the driver/kmod mismatch, they may refuse to do it.

hhue13, 2024-06-21:
> @hhue13 At this point I would try removing the NVIDIA packages, and if you have it, the kmod-kvdo packages as well. On my test machine removing the kmod-kvdo driver allowed normal patching to occur [...]

Thanks for the hint on this one! Based on the attached messages in the dmesg 
[image] I have actually tried to switch the nvidia-driver module to 550 (`dnf module switch-to nvidia-driver:550` which caused that dnf started to downgrade the nvidia packaged and not at least GNOME comes up again.

What I'm wondering here is why was that step required? So far yum installed to proper version of the drivers matching the kernel automatically. Why isn't that working again? Wondering what will happen after the next kernel update.

chindokae, 2024-06-21:
After doing updates on the RHEL8, EPEL, and CUDA repos, all the other systems were successfully updated.  Root cause it hard to determine, but if I had to guess, I'd say that when using modules they all have to work or none of them will work as far as yum or dnf is concerned.   kmods save time and make getting the CUDA stuff right possible, but they have to be perfect.  I've seen this once before and it that case removing the NV driver got rid of the problem.  Every month from then to last month worked.  The anomaly did not persist.

scaronni-nvidia, 2024-06-21:
> I'm running on a Lenovo P16 Gen1 Laptop and that used to work without any issues for over 2 years and broke only with the RHEL 8_10 Upgrade. Since then I can't get this working.
> So it seems that there is still an inconsistency of the driver and the kernel I'm, running. How can I get that fixed?

At the beginning of this thread there was an issue with the metadata for the DNF module streams, where all the kmod packages (precompiled and DKMS) where out of the DNF module streams, leading to unpredictable results and confusion on what gets installed. Somehow it used to work as long as dependencies for `nvidia-kmod` were satisfied. By putting the proper metadata in place (`kmod` packages available in their respective streams) some of this situations popped out.

In your particular case, you have a conflicting packages installed now, Just removed them and continue installing the DNF module stream of your choice.

For example:
```
rpm -qa kmod-nvidia\*
```
Then remove all the results:

```
rpm -e --nodeps <package1> <package2>
```

And keep on installing the module stream of your choice (`dnf module install nvidia-driver:latest-dkms` based on your previous comment).

michaelbarkdoll, 2024-08-07:
I tried uninstalling the driver and reinstalling.  For some reason, the driver doesn't working after being reinstalled.  I can see that `dnf module list nvidia-driver` is defaulting to the open-dkms stream for some reason.  Is there anyway I can correct this to use the latest stream on `dnf install module nvidia-driver:latest` or is the driver still having issues on kernel 4.18.0-553.8.1?

```
[root@host ~]# dnf module list nvidia-driver
Updating Subscription Management repositories.
Last metadata expiration check: 0:03:06 ago on Wed 07 Aug 2024 02:50:50 PM CDT.
cuda-rhel8-x86_64
Name                                   Stream                                 Profiles                                          Summary                                                      
nvidia-driver                          latest [e]                             default [d], fm, ks, src                          Nvidia driver for latest branch                              
nvidia-driver                          latest-dkms                            default [d], fm, ks                               Nvidia driver for latest-dkms branch                         
nvidia-driver                          open-dkms [d]                          default [d], fm, ks, src                          Nvidia driver for open-dkms branch    
```

closed 2024-08-30 by kmittman