NVIDIA/yum-packaging-precompiled-kmod #52: Missing 560 kmods

kmittman, 2024-08-07:
I tried uninstalling the driver and reinstalling.  For some reason, the driver isn't working after being reinstalled.  I can see that `dnf module list nvidia-driver` is defaulting to the open-dkms stream for some reason.  Is there anyway I can correct this to use the latest stream on `dnf install module nvidia-driver:latest` or is the driver still having issues on kernel 4.18.0-553.8.1?

```
[root@host ~]# dnf module list nvidia-driver
Updating Subscription Management repositories.
Last metadata expiration check: 0:03:06 ago on Wed 07 Aug 2024 02:50:50 PM CDT.
cuda-rhel8-x86_64
Name                                   Stream                                 Profiles                                          Summary                                                      
nvidia-driver                          latest [e]                             default [d], fm, ks, src                          Nvidia driver for latest branch                              
nvidia-driver                          latest-dkms                            default [d], fm, ks                               Nvidia driver for latest-dkms branch                         
nvidia-driver                          open-dkms [d]                          default [d], fm, ks, src                          Nvidia driver for open-dkms branch    
```

_Originally posted by @michaelbarkdoll in https://github.com/NVIDIA/yum-packaging-precompiled-kmod/issues/50#issuecomment-2274248542_

kmittman, 2024-08-07:
Hi @michaelbarkdoll we currently do not have 560 kmod packages available. 

We are tracking the issue internally for both [rhel8](https://developer.download.nvidia.com/compute/cuda/repos/rhel8/x86_64/precompiled/) and [rhel9](https://developer.download.nvidia.com/compute/cuda/repos/rhel9/x86_64/precompiled/). Sorry for the inconvenience. 

For now, please use `sudo dnf module install nvidia-driver:555` to install the 555.42.06 precompiled proprietary driver

michaelbarkdoll, 2024-08-07:
Thanks that appears to have fixed the issue.  Was thinking of trying that next.  Sorry, long drive to airport this morning brain still in left field.  Wish you luck with your issue.

kmittman, 2024-08-16:
RHEL 9 kmods are now available for 560, driver `560.28.03` @ kernel `5.14.0-427.31.1`
RHEL 8 kmods should be available in a few hours.

closed 2024-08-21 by kmittman