NVIDIA/yum-packaging-precompiled-kmod #56: kmod-nvidia-550.127.05 for RHEL8 kernels missing

Jojuh, 2024-11-05:
nvidia-drivers 550.127.05 patches vulnerabilities discovered in October. While the drivers are already available, kmod package for that version of drivers is missing for RHEL8, which prohibits the installation of the patched drivers. RHEL9 already seems to have the updated kmod package since 10/31. Is RHEL8 support already dropped? 

Current output of `dnf nvidia-plugin`:
```
Installed kernel(s):
  kernel-core-4.18.0-553.22.1.el8_10.x86_64
  kernel-core-4.18.0-513.24.1.el8_9.x86_64
  kernel-core-4.18.0-513.9.1.el8_9.x86_64

Installed kmod(s):
  kmod-nvidia-550.90.12-4.18.0-553.22.1-3:550.90.12-3.el8_10.x86_64

Available kernel(s):
  kernel-core-4.18.0-553.16.1.el8_10.x86_64
  kernel-core-4.18.0-553.22.1.el8_10.x86_64
  kernel-core-4.18.0-553.5.1.el8_10.x86_64
  kernel-core-4.18.0-553.8.1.el8_10.x86_64
  kernel-core-4.18.0-553.el8_10.x86_64

Available driver(s):
  nvidia-driver-3:550.127.05-1.el8.x86_64
  nvidia-driver-3:550.54.14-1.el8.x86_64
  nvidia-driver-3:550.54.15-1.el8.x86_64
  nvidia-driver-3:550.90.07-1.el8.x86_64
  nvidia-driver-3:550.90.12-1.el8.x86_64
  nvidia-driver-cuda-3:550.127.05-1.el8.x86_64
  nvidia-driver-cuda-3:550.54.14-1.el8.x86_64
  nvidia-driver-cuda-3:550.54.15-1.el8.x86_64
  nvidia-driver-cuda-3:550.90.07-1.el8.x86_64
  nvidia-driver-cuda-3:550.90.12-1.el8.x86_64

Available kmod(s):
  kmod-nvidia-550.54.14-4.18.0-513.18.1-3:550.54.14-3.el8_9.x86_64
  kmod-nvidia-550.54.15-4.18.0-513.18.1-3:550.54.15-3.el8_9.x86_64
  kmod-nvidia-550.54.15-4.18.0-513.24.1-3:550.54.15-3.el8_9.x86_64
  kmod-nvidia-550.90.07-4.18.0-553.16.1-3:550.90.07-3.el8_10.x86_64
  kmod-nvidia-550.90.07-4.18.0-553.5.1-3:550.90.07-3.el8_10.x86_64
  kmod-nvidia-550.90.07-4.18.0-553.8.1-3:550.90.07-3.el8_10.x86_64
  kmod-nvidia-550.90.12-4.18.0-553.22.1-3:550.90.12-3.el8_10.x86_64

Excluded kernel packages during update (4.18.0-553.el8_10):
  bpftool-4.18.0-553.el8_10.x86_64
  kernel-4.18.0-553.el8_10.x86_64
  kernel-abi-stablelists-4.18.0-553.el8_10.noarch
  kernel-core-4.18.0-553.el8_10.x86_64
  kernel-cross-headers-4.18.0-553.el8_10.x86_64
  kernel-debug-4.18.0-553.el8_10.x86_64
  kernel-debug-core-4.18.0-553.el8_10.x86_64
  kernel-debug-devel-4.18.0-553.el8_10.x86_64
  kernel-debug-modules-4.18.0-553.el8_10.x86_64
  kernel-debug-modules-extra-4.18.0-553.el8_10.x86_64
  kernel-devel-4.18.0-553.el8_10.x86_64
  kernel-doc-4.18.0-553.el8_10.noarch
  kernel-headers-4.18.0-553.el8_10.x86_64
  kernel-modules-4.18.0-553.el8_10.x86_64
  kernel-modules-extra-4.18.0-553.el8_10.x86_64
  kernel-tools-4.18.0-553.el8_10.x86_64
  kernel-tools-libs-4.18.0-553.el8_10.x86_64
  perf-4.18.0-553.el8_10.x86_64
  python3-perf-4.18.0-553.el8_10.x86_64
  kernel-tools-libs-devel-4.18.0-553.el8_10.x86_64
```

kmittman, 2024-11-05:
Hi @Jojuh 
We are experiencing test automation issues with the RHEL8 kmod pipelines, aiming for this can be resolved soon.

mfruhner, 2024-11-07:
Hi @kmittman,
does this apply to #55 as well?
Thanks!

kmittman, 2024-11-09:
Hi @Jojuh  and @mfruhner, we have published

| build date	|	package	        |	driver	       |	kernel             	|	tag                    |
|------------------|------------------------|-----------------------|-------------------------------|----------------------------
|2024-11-04	|	kmod-nvidia	|	565.57.01	|	4.18.0-553.27.1	|	el8_10.x86_64 |
|2024-11-04	|	kmod-nvidia	|	560.35.03	|	4.18.0-553.27.1	|	el8_10.x86_64 |
|2024-11-04	|	kmod-nvidia	|	550.127.05	|	4.18.0-553.27.1	|	el8_10.x86_64 |
|2024-11-04	|	kmod-nvidia	|	535.216.01	|	4.18.0-553.27.1	|	el8_10.x86_64 |

closed 2024-11-09 by kmittman