NVIDIA/yum-packaging-precompiled-kmod #57: Missing prebuilt kmod for RHEL 9.5 5.14.0-503.23.2.el9_5.x86_64

droghio, 2025-02-19:
Due to a security patch a new RHEL kernel was pushed about a week back: https://access.redhat.com/errata/RHSA-2025:1262

I don't see a corresponding kernel version in the main page table. Just checking if there was a new build on the way.

[image]

chindokae, 2025-02-20:
Redhat has released another kernel.  The current version on their repo is 5.14.0.503.26.1.el9_5

kmittman, 2025-02-21:
Hi there @droghio and @chindokae, sorry about the delay -
The `535`, `550`, `565` branch precompiled proprietary kmods for `5.14.0-503.26.1` kernel are now available.

There is a bug in `nvidia-kmod-common-570.86.15-1` that is not properly blacklisting the nouveau module, this was blocking CI from passing; it will be fixed in the next 570 driver release. Until then, prebuilt `570` kmods will unfortunately not be available.

chindokae, 2025-02-21:
Thanks for getting this out and thanks for catching that bug - the folks
running the APT repos didn't.

neosr-project, 2025-02-21:
Hi, thanks for explaining the reason. Any ETA from the nvidia team on 570 builds for RHEL?
Related issue: https://github.com/NVIDIA/yum-packaging-nvidia-plugin/issues/16

droghio, 2025-02-21:
I think we should probably keep tracking the 570 release in that other bug just to keep things easier to understand. Was someone able to get the nvidia driver loaded? I tried on my machine and although the cuda driver installed fine I wasn't able to find the X11 driver for 56 (X is falling back to software render although nvidia-smi works...). If someone else got things to work alright I'll assume its just something messed up on my side and close the ticket.

Thanks for the quick response btw!

droghio, 2025-02-26:
Finally got it working. For some reason needed to manually `modprobe nvidia-drm` for it to take (even after a few reboots) but display manager now works as expected. Thanks for the fix!

closed 2025-02-26 by droghio

kmittman, 2025-03-04:
FYI @droghio, @chindokae, @neosr-project the kmod for `570` branch stream is now available, specifically driver `570.124.06` @  `5.14.0-503.26.1.el9_5.x86_64` kernel