NVIDIA/yum-packaging-precompiled-kmod #58: KMODS for RHEL8 and RHEL9 need updating.

chindokae, 2025-05-19:
Kernels for RHEL8 and RHEL9 have been updated.   I am not sure when it updated but I checked last Thursday and this was the status:

RHEL 8 needs to be updated to x86_64 4.18.0-553.52.1.el8_10. Current kmods:            4.18.0-553.51.1 (4/29/25)

RHEL 9 needs to be updated to x86_64 5.14.0-570.17.1.el9_6    Current kmods:   	5.14.0-503.40.1 (4/29/25)

kmittman, 2025-05-19:
Hi @chindokae thank you, we are looking into this

chindokae, 2025-05-22:
Hi again and I hate to be a bother, but is there any word on this?   Whether it was intended or not, these modules have become a key pacing item in enterprise patching processes as we have added CUDA workloads.   Until there are modules, none of the systems can be patched, and Cyber knows we are "late" again.   It would be very helpful if we could respond to their inquiries with an estimated date that we will be able to push patches.   Is there one?

kmittman, 2025-05-22:
RHEL9 kmods packages for kernel `5.14.0-570.17.1` are now available.
RHEL8 pipeline automation is still blocked, working on getting this out soon.

kmittman, 2025-05-23:
Okay RHEL9 kmod packages for kernel `4.18.0-553.53.1` are now available 
> note: the status page is stale, working on that
```
echo module_hotfixes=1 >> /etc/yum.repos.d/cuda-rhel8.repo
dnf nvidia-plugin | grep 4.18.0-553.53.1 | sort -Vr
  kmod-nvidia-575.51.03-4.18.0-553.53.1-3:575.51.03-3.el8_10.x86_64
  kmod-nvidia-570.133.20-4.18.0-553.53.1-3:570.133.20-3.el8_10.x86_64
  kmod-nvidia-550.163.01-4.18.0-553.53.1-3:550.163.01-3.el8_10.x86_64
  kmod-nvidia-535.247.01-4.18.0-553.53.1-3:535.247.01-3.el8_10.x86_64
sudo sed -i '/module_hotfixes/d' /etc/yum.repos.d/cuda-rhel8.repo
```
Regarding the SLA concerns, feel free to send me an email.

closed 2025-08-21 by kmittman

simtheverse, 2025-08-22:
Thank you! Works great!