NVIDIA/yum-packaging-precompiled-kmod #60: kmod-nvidia-575.57.08 missing from the precompiled repo

lfarkas, 2025-06-16:
all package for the version 575.57.08 are present in the repo:
https://developer.download.nvidia.com/compute/cuda/repos/rhel9/x86_64/
except the precompiled kmod package itself 
kmod-nvidia-575.57.08-5.14.0-570.18.1-575.57.08-1.el9_6.x86_64.rpm
is it just a bug or it's failed to build?
of course the dkms and  present but would be useful to add that missing one package.

kmittman, 2025-06-16:
On Tuesday, we had a RC with a precompiled kmod for driver `575.57.08`  and kernel `5.14.0-570.21.1` however the test CI failed. Will take a look, thanks.

FYI, the  dashboard is here: https://developer.download.nvidia.com/compute/cuda/repos/rhel9/x86_64/precompiled/

lfarkas, 2025-06-16:
ok but i write about the 575.57.08 and kernel 5.14.0-570.18.1 which has all components in the repo except the kmod itself.

kmittman, 2025-06-27:
I see kmods available for kernel `5.14.0-570.23.1` now

lfarkas, 2025-06-27:
yes thanks

closed 2025-06-27 by lfarkas

lfarkas, 2025-06-27:
but not working with NVIDIA GeForce RTX 5060 Ti.
https://www.nvidia.com/en-in/drivers/results/ show for this card:
570.169      Release Date: Tue Jun 17, 2025
570.153.02 Release Date: Mon May 19, 2025
nut none of these are in the precompiled cuda drivers. and only the open dkms driver works with this card.